jerryscript-project/jerryscript #2258: stack-overflow in ecma_builtin_json_array

renatahodovan, 2018-03-28:
###### Jerry version:

```
Checked revision: 79289714
Build command: ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --compile-flag=-g --jerry-libc=off --static-link=off --strip=off --system-allocator=on --linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```
###### OS:

```
Ubuntu 17.10
```
###### Test case:

```javascript
JSON.stringify(10, Array)
```

Usually, stack-overflow issues happen due to recursive js calls which is expected.
However, this test case doesn't contain any explicit recursion, so it may uncover a real issue.

###### Backtrace:

```
ASAN:DEADLYSIGNAL
=================================================================
==30252==ERROR: AddressSanitizer: stack-overflow on address 0xff6cfee0 (pc 0xf7a9eb38 bp 0xff6d0308 sp 0xff6cfec0 T0)
    #0 0xf7a9eb37 in malloc (/usr/lib32/libasan.so.4+0xe4b37)
    #1 0x566a417b in jmem_heap_alloc_block_internal jerryscript/jerry-core/jmem/jmem-heap.c:324
    #2 0x566a4248 in jmem_heap_gc_and_alloc_block jerryscript/jerry-core/jmem/jmem-heap.c:360
    #3 0x566a431d in jmem_heap_alloc_block jerryscript/jerry-core/jmem/jmem-heap.c:406
    #4 0x56680b7d in ecma_alloc_property_pair jerryscript/jerry-core/ecma/base/ecma-alloc.c:240
    #5 0x566c3138 in ecma_create_property jerryscript/jerry-core/ecma/base/ecma-helpers.c:452
    #6 0x566c3809 in ecma_create_named_data_property jerryscript/jerry-core/ecma/base/ecma-helpers.c:538
    #7 0x5669836b in ecma_op_general_object_define_own_property jerryscript/jerry-core/ecma/operations/ecma-objects-general.c:353
    #8 0x566d4e80 in ecma_op_array_object_define_own_property jerryscript/jerry-core/ecma/operations/ecma-array-object.c:313
    #9 0x5669bfed in ecma_op_object_define_own_property jerryscript/jerry-core/ecma/operations/ecma-objects.c:1133
    #10 0x566cd821 in ecma_builtin_helper_def_prop jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-helpers.c:682
    #11 0x566d44f0 in ecma_op_create_array_object jerryscript/jerry-core/ecma/operations/ecma-array-object.c:120
    #12 0x5668befb in ecma_builtin_array_dispatch_construct jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-array.c:101
    #13 0x5668be94 in ecma_builtin_array_dispatch_call jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-array.c:87
    #14 0x566d0679 in ecma_builtin_dispatch_call jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:915
    #15 0x5669277b in ecma_op_function_call jerryscript/jerry-core/ecma/operations/ecma-function-object.c:439
    #16 0x566582b7 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1331
    #17 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #18 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #19 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #20 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #21 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #22 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #23 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #24 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #25 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #26 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425

    ...

    #227 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #228 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #229 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #230 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #231 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #232 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #233 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #234 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #235 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #236 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #237 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #238 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #239 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #240 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #241 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #242 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #243 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #244 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #245 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #246 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #247 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688
    #248 0x56658840 in ecma_builtin_json_str jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1425
    #249 0x56659a85 in ecma_builtin_json_array jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.c:1688

SUMMARY: AddressSanitizer: stack-overflow (/usr/lib32/libasan.so.4+0xe4b37) in malloc
==30252==ABORTING
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) </sup>

rerobika, 2019-01-16:
I think it is not a real bug, it's only an infinity recursion due the replacer function (`Array`).

I tested this issue in:
 - Chrome:
```
JSON.stringify(10, Array)
VM236:1 Uncaught RangeError: Maximum call stack size exceeded
    at JSON.stringify (<anonymous>)
    at <anonymous>:1:6
```
- Firefox
```
JSON.stringify(10, Array)
InternalError: too much recursion
```

@LaszloLango can we close this issue?

akosthekiss, 2019-01-16:
Note: Both referenced engines catch the problem of recursion themselves and terminate the JS without crashing.

PR #2969 (2019-07-23): Rework the engine's internal recursion limit
This patch unifies the recursion limit checking for RegExp, function call and JSON as well.
Until now the limit was only a counter which was increased/decreased at certain points.
This counter has been substituted with a numeric limit which allows to restrict the stack usage.

This patch fixes #2963 and resolves the closed #2258 issue.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 4a9e1858 (2019-07-23): Rework the engine's internal recursion limit (#2969)

This patch unifies the recursion limit checking for RegExp, function call and JSON as well.
Until now the limit was only a counter which was increased/decreased at certain points.
This counter has been substituted with a numeric limit which allows to restrict the stack usage.

This patch fixes #2963 and resolves the closed #2258 issue.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu