jerryscript-project/jerryscript #2268: Suggest caching Zephyr SDK for Travis speed / success improvement

grgustaf, 2018-04-03:
This [Travis build](https://travis-ci.org/jerryscript-project/jerryscript/jobs/361777717) failed for me while trying to download the Zephyr SDK. I wonder if you're seeing that a lot? I used to have that problem in ZJS but I solved it with Travis caching in this [pull request](https://github.com/intel/zephyr.js/commit/ce313765b829108a308edaf675d390752f29c80f).

If it had been that simple here I would have tried it and submitted a PR right away, but it looks like you'd need to edit `zephyr/Makefile.travis` as well as `.travis.yml` and I don't have time at the moment to look into this, so I wanted to make a note of it.

This also has the benefit of skipping the download most of the time, speeding up the Travis build a lot. So basically, if you do it like I did, you'd cache some directory that holds the SDK and then check to see if it already contains the right version before you bother to download it with `wget` in the Makefile.

I'd also suggest parameterizing the Zephyr build version like we've done in ZJS to make it easier to update.

I intend to look at this someday myself but in case someone cares to beat me to it, here you go. :)

LaszloLango, 2018-04-12:
A new error occurred on Zephyr job of Travis CI:
> setuptools_scm.version.SetuptoolsOutdatedWarning: your setuptools is too old (<12)

akosthekiss, 2018-05-18:
Is this still an issue? #2333 has solved the build error and the build time is under 5 mins for Zephyr.