jerryscript-project/jerryscript #2283: Add native ffi support in jerryscript.

lygstate, 2018-04-16:


lygstate, 2018-04-17:
@martijnthe That's very good:), But I am talking about ffi:) for example:
https://github.com/node-ffi-napi/node-ffi-napi
These are different topic:)

akosthekiss, 2018-04-26:
@lygstate The project you've linked is a Node.js addon. JerryScript is not the lightweight equivalent of Node.js. It is "just" an engine. You can rather compare it to the V8 engine that's driving Node.js. Please, consider whether a project that uses JerryScript might be a better fit for your proposal (e.g., IoT.js).

lygstate, 2018-04-27:
use same API for binding create， so we could be able to use a lot of npmjs project

akosthekiss, 2018-04-27:
@lygstate Please note that JerryScript was _never_ intended to (re)use Node.js packages. Samsung/iotjs is a platform on top of JerryScript, that project has goals that you are mentioning.

akosthekiss, 2018-05-10:
Just double checked the referenced project and ffi in general. Although there might be some cases (which I haven't thoroughly elaborated) where libffi, the C library behind node-ffi-napi might be useful, the whole concept of node-ffi-napi is based on dynamic library loading (dlopen), dynamic symbol retrieval (dlsym), etc., which are concepts unknown and alien to the JerryScript engine. So, I'm closing this as out of scope.

Should anyone feel otherwise or even think that it can be implemented, e.g., based on jerry-ext/{arg,module}, then do comment on and/or reopen the issue. (Of course, a working prototype in a PR always helps making a stronger case...)

closed 2018-05-10 by akosthekiss

jan4984, 2019-04-29:
I know JerryScript will not have native FFI. Just ask here is any project implement the JerryScript FFI?

legendecas, 2019-04-29:
@jan4984 https://github.com/shadow-node/ffi is a simple ffi made for shadow-node with jerryscript API.

jan4984, 2019-05-06:
@legendecas That's good. The problem is I don't need neither want ShadowNode.