jerryscript-project/jerryscript #2284: Using jerryscript in Mobile Devices

yemingyu, 2018-04-17:
I want to use jerryscript in iOS and Android, can I use jerryscript through referencing the source code?

akosthekiss, 2018-04-17:
@yemingyu JerryScript may be freely used according to its license (Apache 2.0, which is quite permissive, you only have to mention the original license and copyright notices in derivative works -- roughly). Regarding referencing the source code, that's the usual way of using the project. There are no officially prepackaged or prebuilt libs of the engine, but other projects typically use and build the sources of JerryScript directly, e.g., as a git submodule in their own source tree.

Let us know if this answers your question (and whether it suites your needs).

yemingyu, 2018-04-19:
I don't know what part jerryscript needs to refer to, whether there is a document or a c project example？

yemingyu, 2018-04-19:
OK，I will try it，thank you very much!

closed 2018-04-19 by yemingyu