jerryscript-project/jerryscript #2297: How to provide more error information?

BernardXiong, 2018-04-24:
In the Unix build, if there are some errors in script, only little message provided:
```JavaScript
A();
```

    ./bin/jerry error.js
    Script Error: ReferenceError

However, in iot.js, more error message provided:

    ./build/x86_64-linux/debug/bin/iotjs error.js 
    uncaughtException: ReferenceError: A is not defined

How to show this message? Thanks.

LaszloLango, 2018-04-24:
You must turn on the error messages. They are disabled by default to save some ROM space.

`tools/build.py --error-messages=on`

Furthermore there is a new feature which recently landed into the master branch. Now you can get a backtrace too if you enable the `line-info`.

You can check the build options for details: `tools/build.py --help`
```
--error-messages X    enable error messages (ON, OFF; default: OFF)
--line-info X         provide line info (ON, OFF; default: OFF)
```

BernardXiong, 2018-04-24:
Thank @LaszloLango 

I put JerryScript in RT-Thread as a standalone package:
https://github.com/RT-Thread-packages/jerryscript

The scons is used in RT-Thread. Therefore, I have to put a macro-definition in compiler command by myself. If I only define -DJERRY_FEATURE_ERROR_MESSAGES, it's not working. What's the correct way?

LaszloLango, 2018-04-24:
For CMake users the following options must be used: `-DFEATURE_ERROR_MESSAGES=ON` and `-DFEATURE_LINE_INFO=ON`

zherczeg, 2018-04-24:
Wrong macro name. It should be `JERRY_ENABLE_ERROR_MESSAGES`

BernardXiong, 2018-04-24:
Done! Thank you very much.

closed 2018-04-24 by BernardXiong