jerryscript-project/jerryscript #2301: jerry_cleanup -- quiet, please...

t-harvey, 2018-04-26:
Folks,
When I am done using the interpreter, I would like a way to clean up (in case I want to start a new instance of the interpreter, for example) that doesn't complain about un-free'd jerry_value_t values.

It's useful sometimes to have jerry_cleanup complain to tell me when I've got a leak, but there are times when I just want it to clean up everything and exit, without lecturing :-) me. 

...is this possible now, can we add a flag to jerry_cleanup ("quiet"), or is there some deep technical issue I don't understand that precludes this functionality?

Thanks,
Tim

As an example, I'd like to write code like:

main()
{
   for (i = 0; i < forever; i++)
   {
      jerry_init();
      ...do some stuff, leak pointers/memory all over the interpreter, etc.
      jerry_cleanup("quiet"); /* reset the world sans error messages/exceptions/etc. */
   }
} /* main */

zherczeg, 2018-04-26:
Probably the easy way is calling jerry_init() again, although that might cause some assertion fails. The ugly part is that native objects will never be freed in this case, and resources such as sockets may kept until the application exits. Perhaps we could add a jerry_reset() which frees native objects and resets state.

martijnthe, 2018-04-26:
Why bother calling jerry_cleanup at all if you’re not interested in properly cleaning up?

t-harvey, 2018-04-26:
Martijn,
First, I'm a fairly naive user -- if the docs say to call jerry_obtuse_function(), I call it and let it do its voodoo... :-)

Secondly, I _would_ like JS to clean up after me -- I sometimes create a jerry_value_t thing and then lose track of it, but I don't really care: I just want to have some kind of barrier beyond which I know the interpreter has been reset.

To expand upon my example; imagine that I want to print out the result of some invocation of the interpreter and then kill the interpreter -- it's perfectly natural to ignore the result of the eval (and I'd like to do so without being yelled at :-) :
```
main()
{
   for (i = 0; i < forever; i++)
   {
      jerry_init();
      ...compute the result of some computation, and at the last minute:
      jerry_eval ("print(result)", 13, false);
      jerry_cleanup("quiet"); /* reset the world sans error messages/exceptions/etc. */
   }
} /* main */
```
...other times, I might create some long-lived global objects.  In these structures, I've taken to storing each one's jerry_value_t in itself to keep it from being garbage collected, and then I can forget my own handle to it, looking it up by name whenever I need to use it.  So at the end of my computation, I've got a bunch of these global objects which I would otherwise have to keep my own set of pointers to, and I simply want them to cleanly (and quietly :-) go away.

Judging from yours and Zoltan's comments, the "quiet" option doesn't exist; the question, then, is: can it (please)?

Indeed, on re-reading the above, it occurs to me:
1) assume that jerry_cleanup does what it says: shuts the interpreter down, cleaning up all memory
2) then the default behavior should be a quiet cleaning up of all memory, and the exceptional (currently: only) behavior should be to warn the user that he hasn't carefully kept track of every Javascript value he's created (for cases where the user actually cares, wants to track down bugs, etc.).

Thanks,
Tim

rerobika, 2019-10-25:
I'm highly against this idea, and IMO this should not be added to the public API. This change would soften the restrictions of reference management in the API which can lead to/cause undefined behaviors.

> Secondly, I would like JS to clean up after me --

The API is designed for releasing/acquire `jerry_value`s and the user's responsibility to manage these references.

>   I sometimes create a jerry_value_t thing and then lose track of it, but I don't really care: 

How about using the [auto release](https://jerryscript.net/ext-reference-autorelease/) feature?

Also as @martijnthe suggested the `deinit_cb` and `finalize_cb` hooks can help to release values.

I'm closing the issue with this conclusion and the lack of inactivity, please reopen it anytime if needed.

closed 2019-10-25 by rerobika