jerryscript-project/jerryscript #2302: instanceof throws TypeError for instance derived from native-backed constructor

gabrielschulhof, 2018-04-28:
```C
#include <stdio.h>
#include "jerryscript.h"

static const char instanceof_source[] = "var x = function(o, c) {return (o instanceof c);}; x";

static jerry_value_t
external_function (const jerry_value_t function_obj,
                   const jerry_value_t this_arg,
                   const jerry_value_t args_p[],
                   const jerry_size_t args_count)
{
  (void) function_obj;
  (void) this_arg;
  (void) args_p;
  (void) args_count;

  return jerry_create_undefined ();
} /* external_function */

static void
test_instanceof (const char *prefix,
                 jerry_value_t undefined,
                 jerry_value_t instanceof,
                 jerry_value_t constructor)
{
  jerry_value_t instance = jerry_construct_object (constructor, NULL, 0);
  jerry_value_t args[2] =
  {
    instance, constructor
  };

  jerry_value_t result = jerry_call_function (instanceof, undefined, args, 2);
  if (jerry_value_has_error_flag (result))
  {
    jerry_value_clear_error_flag (&result);
  }

  jerry_value_t js_string = jerry_value_to_string (result);
  jerry_size_t size = jerry_get_string_size (js_string);
  jerry_char_t c_string[size + 1];
  jerry_string_to_char_buffer (js_string, c_string, size);
  c_string[size] = 0;
  fprintf(stderr, "%8s: instanceof results in: %s\n", prefix, c_string);

  jerry_release_value (instance);
  jerry_release_value (js_string);
  jerry_release_value (result);
}

int
main (void)
{
  jerry_init (JERRY_INIT_EMPTY);

  jerry_value_t constructor;

  jerry_value_t undefined = jerry_create_undefined ();
  jerry_value_t instanceof = jerry_eval ((jerry_char_t *) instanceof_source, sizeof (instanceof_source) - 1, true);

  /* Test for a native-backed function. */
  constructor = jerry_create_external_function (external_function);

  test_instanceof ("native", undefined, instanceof, constructor);
  jerry_release_value (constructor);

  /* Test for a JS constructor. */
  jerry_value_t global = jerry_get_global_object ();
  jerry_value_t object_name = jerry_create_string ((jerry_char_t *) "Object");
  constructor = jerry_get_property (global, object_name);
  jerry_release_value (object_name);
  jerry_release_value (global);

  test_instanceof ("js", undefined, instanceof, constructor);
  jerry_release_value (constructor);

  jerry_release_value (instanceof);
  jerry_release_value (undefined);

  jerry_cleanup ();
} /* main */
```

LaszloLango, 2018-05-24:
Hi Gabriel, sorry for the late answer. IMHO this is a bug and it should work. @zherczeg is this intentional or is it a bug?

akosthekiss, 2019-06-28:
This issue seems to have unanswered questions still.

@zherczeg Could you please follow up on this?

rerobika, 2019-10-22:
Hi @gabrielschulhof!

Sorry for the late response, that's a valid issue since this should definitely work. I've just submitted a PR (#3246) to resolve this problem. By far, now you can use [jerry_binary_operation](https://github.com/jerryscript-project/jerryscript/blob/master/docs/02.API-REFERENCE.md#jerry_binary_operation), with `JERRY_BIN_OP_INSTANCEOF` first argument to perform `instanceof` operation from the API.

PR #3246 (2019-10-25): Fix lazy instantiation of external functions 'prototype' property
This patch resolves #2302.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit b7aa21eb (2019-10-25): Fix lazy instantiation of external functions 'prototype' property (#3246)

This patch resolves #2302.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2019-10-25 by rerobika