jerryscript-project/jerryscript #2310: Passing Error objects to Promise

AnthonyCalandra, 2018-05-04:
Hello,

I'm writing code that should work as the following from the JS side:
```js
x.y.fn('foo')
  .then(v => console.log(v))
  .catch(error => console.err(err)); 
```
Don't worry about what happens when the promise is resolved, my problem is when I reject the promise. Currently, I want to reject the promise and pass an error object to the catch handler when the supplied string to `fn` is not a special value.

The code I have looks like the following:
```c
  jerry_value_t returned_promise = jerry_create_promise();
  if (string is not correct) {
    jerry_value_t error = error_argument_invalid_at_index(0, "Unrecognized algorithm");
    jerry_value_t result = jerry_resolve_or_reject_promise(returned_promise, error, false);
    DEBUG_LOG("RESULT STATUS: %d", jerry_get_error_type(result));
    // If something went wrong rejecting the promise, error out.
    if (jerry_value_has_error_flag(result)) {
      jerry_release_value(returned_promise);
      jerry_release_value(error);
      return result;
    }

    // No issue rejecting the promise so return it.
    jerry_release_value(result);
    jerry_release_value(error);
    return returned_promise;
  }

   // ...
```
The function `error_argument_invalid_at_index` is a utility function that creates an `jerry_value_t` using `jerry_create_error`. When I run my JS code I get the following error:
```
ICE: Unreachable control path at .../vendor/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c(ecma_copy_value):Error: ERR_FAILED_INTERNAL_ASSERTION
Fatal Jerryscript Error: ERR_FAILED_INTERNAL_ASSERTION
```

Am I not allowed to pass an error object in? Documentation doesn't seem to say. Any help would be appreciated! Thanks.

galpeter, 2018-05-04:
Hi,
Error objects are a bit special, they are objects and have the error flag set. It seems there is a missing handler for this case in the promises (note, I did not checked this, just based on the error message).

Could you try calling the ` jerry_value_clear_error_flag(&error);`  code before passing the error object for the reject/resolve? This could be a workaround before we check the report in more detail.

AnthonyCalandra, 2018-05-04:
Calling `jerry_value_clear_error_flag(&error);` right after creating the error object and passing that into `jerry_resolve_or_reject_promise` didn't seem to do anything. By this I mean there was no assertion failure, but my rejection handler also didn't print any output (so I'm assuming it's not getting called at all). For the app I'm working on it's acceptable for me to pass a string down instead so I'll be fine.

AnthonyCalandra, 2018-05-25:
Today I cloned latest master and wrote a unit test to see if I could capture this behavior in order to investigate it, but noticed the test passed! You can see the test here: https://github.com/AnthonyCalandra/jerryscript/blob/920b5bd7a5a22bea321104dc4be6975a172045fb/tests/unit-core/test-promise-error-object.c

Thinking this issue might have been resolved now?

closed 2018-06-25 by AnthonyCalandra