jerryscript-project/jerryscript #2365: LCACHE miss every time calling native C function

jack9766, 2018-05-29:
I have noticed everytime JS calls an external C function the LCACHE will miss.

Testing method:
I have loop a external C call 10000 times and LCACHE missed exactly 20000 times.

Is there a way to disable lcache right before external function calls?

jack9766, 2018-05-29:
Here's how I tested it:

I slightly modified the external function example:

```
#include <string.h>
#include "jerryscript.h"
#include "jerryscript-ext/handler.h"

/**
 * Add param to 'this.x'
 */
static jerry_value_t
add_handler (const jerry_value_t func_value, /**< function object */
             const jerry_value_t this_val, /**< this arg */
             const jerry_value_t *args_p, /**< function arguments */
             const jerry_length_t args_cnt) /**< number of function arguments */
{
  return jerry_create_undefined ();
} /* add_handler */

int
main (void)
{
  /* Initialize engine */
  jerry_init (JERRY_INIT_EMPTY);

  /* Register 'print' function from the extensions */
  jerryx_handler_register_global ((const jerry_char_t *) "print",
                                  jerryx_handler_print);

  /* Create a JS object */
  const jerry_char_t my_js_object[] = " \
    MyObject = \
    { x : 12, \
      y : 'Value of x is ', \
      foo: function () \
      { \
        return this.y + this.x; \
      } \
    } \
  ";

  jerry_value_t my_js_obj_val;

  /* Evaluate script */
  my_js_obj_val = jerry_eval (my_js_object,
                              strlen ((const char *) my_js_object),
                              false);

  /* Create a JS function object and wrap into a jerry value */
  jerry_value_t add_func_obj = jerry_create_external_function (add_handler);

  /* Set the native function as a property of previously created MyObject */
  jerry_value_t prop_name = jerry_create_string ((const jerry_char_t *) "add2x");
  jerry_set_property (my_js_obj_val, prop_name, add_func_obj);
  jerry_release_value (add_func_obj);
  jerry_release_value (prop_name);

  /* Free JavaScript value, returned by eval (my_js_object) */
  jerry_release_value (my_js_obj_val);

  const jerry_char_t script[] = "var i; for(i=0;i<100000;i++){MyObject.add2x ();}";
  size_t script_size = strlen ((const char *) script);

  /* Evaluate script */
  jerry_value_t eval_ret = jerry_eval (script, script_size, false);

  /* Free JavaScript value, returned by eval */
  jerry_release_value (eval_ret);

  /* Cleanup engine */
  jerry_cleanup ();

  return 0;
}

```

Now add a static counter to `ecma_lcache_lookup()` in `jerry-core/ecma/base/ecma-lcache.c` that increment everytime it returns NULL and log it out.

Mine test outputed:

`200017 MISSES`

zherczeg, 2018-05-30:
There should be no connection between native functions and lcache. Lcache is designed to store property names, and native functions are callable objects.

jack9766, 2018-05-30:
The miss numbers is directly connected to the number of times the native function is called. I also tested this with a JS function which resulted in only 9 misses for 100000 calls.

jack9766, 2018-05-31:
I have narrowed down the lcache misses:

* Both cache misses are trying to search for the same property, here are the stack traces:
```
#1  0x0000000000411ba4 in ecma_lcache_lookup (object_p=0x66a338 <jerry_global_heap+24>, prop_name_p=0x66a768 <jerry_global_heap+1096>) at jerryscript/jerry-core/ecma/base/ecma-lcache.c:219
#2  0x000000000044d312 in ecma_op_resolve_reference_value (lex_env_p=0x66a348 <jerry_global_heap+40>, name_p=0x66a768 <jerry_global_heap+1096>) at jerryscript/jerry-core/ecma/operations/ecma-reference.c:91
#3  0x000000000043fb3b in vm_loop (frame_ctx_p=0x7fffffffa100) at jerryscript/jerry-core/vm/vm.c:827
#4  0x0000000000443beb in vm_execute (frame_ctx_p=0x7fffffffa100, arg_p=0x7fffffffa2e0, arg_list_len=2) at jerryscript/jerry-core/vm/vm.c:2988
#5  0x0000000000443edb in vm_run (bytecode_header_p=0x66a748 <jerry_global_heap+1064>, this_binding_value=27, lex_env_p=0x66a348 <jerry_global_heap+40>, is_eval_code=false, arg_list_p=0x7fffffffa2e0, arg_list_len=2)
    at jerryscript/jerry-core/vm/vm.c:3081
#6  0x000000000042f165 in ecma_op_function_call (func_obj_p=0x66a7c8 <jerry_global_heap+1192>, this_arg_value=72, arguments_list_p=0x7fffffffa2e0, arguments_list_len=2)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.c:500
#7  0x0000000000403a33 in jerry_invoke_function (is_invoke_as_constructor=false, func_obj_val=1195, this_val=72, args_p=0x7fffffffa2e0, args_count=2) at jerryscript/jerry-core/api/jerry.c:2258
#8  0x0000000000403a80 in jerry_call_function (func_obj_val=1195, this_val=72, args_p=0x7fffffffa2e0, args_count=2) at jerryscript/jerry-core/api/jerry.c:2284
```
```
#1  0x0000000000411ba4 in ecma_lcache_lookup (object_p=0x66a338 <jerry_global_heap+24>, prop_name_p=0x66a768 <jerry_global_heap+1096>) at jerryscript/jerry-core/ecma/base/ecma-lcache.c:219
#2  0x0000000000408817 in ecma_find_named_property (obj_p=0x66a338 <jerry_global_heap+24>, name_p=0x66a768 <jerry_global_heap+1096>) at jerryscript/jerry-core/ecma/base/ecma-helpers.c:614
#3  0x000000000042ce4e in ecma_op_object_find_own (base_value=27, object_p=0x66a338 <jerry_global_heap+24>, property_name_p=0x66a768 <jerry_global_heap+1096>) at jerryscript/jerry-core/ecma/operations/ecma-objects.c:513
#4  0x000000000042d055 in ecma_op_object_find (object_p=0x66a338 <jerry_global_heap+24>, property_name_p=0x66a768 <jerry_global_heap+1096>) at jerryscript/jerry-core/ecma/operations/ecma-objects.c:615
#5  0x000000000044d3ac in ecma_op_resolve_reference_value (lex_env_p=0x66a348 <jerry_global_heap+40>, name_p=0x66a768 <jerry_global_heap+1096>) at jerryscript/jerry-core/ecma/operations/ecma-reference.c:115
#6  0x000000000043fb3b in vm_loop (frame_ctx_p=0x7fffffffa100) at jerryscript/jerry-core/vm/vm.c:827
#7  0x0000000000443beb in vm_execute (frame_ctx_p=0x7fffffffa100, arg_p=0x7fffffffa2e0, arg_list_len=2) at jerryscript/jerry-core/vm/vm.c:2988
#8  0x0000000000443edb in vm_run (bytecode_header_p=0x66a748 <jerry_global_heap+1064>, this_binding_value=27, lex_env_p=0x66a348 <jerry_global_heap+40>, is_eval_code=false, arg_list_p=0x7fffffffa2e0, arg_list_len=2)
    at jerryscript/jerry-core/vm/vm.c:3081
#9  0x000000000042f165 in ecma_op_function_call (func_obj_p=0x66a7c8 <jerry_global_heap+1192>, this_arg_value=72, arguments_list_p=0x7fffffffa2e0, arguments_list_len=2)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.c:500
#10 0x0000000000403a33 in jerry_invoke_function (is_invoke_as_constructor=false, func_obj_val=1195, this_val=72, args_p=0x7fffffffa2e0, args_count=2) at jerryscript/jerry-core/api/jerry.c:2258
#11 0x0000000000403a80 in jerry_call_function (func_obj_val=1195, this_val=72, args_p=0x7fffffffa2e0, args_count=2) at jerryscript/jerry-core/api/jerry.c:2284
```
* the opcode and operands that leads to the caches misses are `42` and `640`.
* In addition I tested with `print` ext-handler and confirmed that it also encounter 2 cache misses per call.

LaszloLango, 2018-06-13:
@jack9766 hi,
I've looked into this. The problem comes from here:
https://github.com/jerryscript-project/jerryscript/blob/efa88507833f554bcf2dc1f901e3549529bb99da/jerry-core/ecma/base/ecma-lcache.c#L204

To make it work you need to add the external function names to the magic string list. See: http://jerryscript.net/api-reference/#jerry_register_magic_strings

zherczeg, 2018-06-13:
Thank you for the analysis. I think we can close this now. Feel free to reopen if needed.

closed 2018-06-13 by zherczeg