jerryscript-project/jerryscript #2366: external function type flag is lost after returning from create_external_function

jack9766, 2018-05-30:
So I was trying to debug my last [issue](https://github.com/jerryscript-project/jerryscript/issues/2365) and I ran into what I think its a bug.

***Explanation***

* `jerry_create_external_function_object` calls [`ecma_op_create_external_function_object `](https://github.com/jerryscript-project/jerryscript/blob/64b16bf190cc1965549c395f5fe9513ef3cd74f4/jerry-core/ecma/operations/ecma-function-object.c#L257).

* If you look at [`ecma_op_create_external_function_object`](https://github.com/jerryscript-project/jerryscript/blob/64b16bf190cc1965549c395f5fe9513ef3cd74f4/jerry-core/ecma/operations/ecma-function-object.c#L257), it sets the type flag of the function object by calling [`ecma_create_object`](https://github.com/jerryscript-project/jerryscript/blob/47794512846d74a1255016ea087ab2003d1f1d6b/jerry-core/ecma/base/ecma-helpers.c#L76) with `ECMA_OBJECT_TYPE_EXTERNAL_FUNCTION`.

* You can check the flag before `ecma_op_create_external_function_object` by logging with `printf("%x\n", function_obj_p->type_flags_refs);` and it will print `0x63`.

* Now `jerry_create_external_function_object` will return the newly created function object and the user will call [`jerry_set_property`](https://github.com/jerryscript-project/jerryscript/blob/ac9fce1d8d5a8eb821525f09ef1607e4643e14f3/jerry-core/api/jerry.c#L1929) to register the external function.

* If you log out the type flag again in [`jerry_set_property`](https://github.com/jerryscript-project/jerryscript/blob/ac9fce1d8d5a8eb821525f09ef1607e4643e14f3/jerry-core/api/jerry.c#L1929) using `printf("%x", ecma_get_object_from_value (obj_value)->type_flags_refs)`, it will log out `0x60`. **So it lost its type flag somehow**.

*I checked the `prop_name_val` by converting it to `char *` then `strcmp` to make sure I was actually logging the newly created external function object.*

galpeter, 2018-05-31:
In the jerry_set_propery call, the `value_to_set` variable is the external function. the `obj_value` is the object to set the property on. Is this what you wanted to log out?

jack9766, 2018-05-31:
You are right, I logged the wrong object.

closed 2018-05-31 by jack9766