jerryscript-project/jerryscript #2368: Side effect of using uncompressed pointer?

jack9766, 2018-06-01:
I was wonder if there are any side effect besides increased memory usage by using uncompressed pointer(passing `--cpointer-32bit` as a build option).

My main concern is according to the [data representation](https://github.com/jerryscript-project/jerryscript/blob/master/docs/04.INTERNALS.md#data-representation) section of the docs:

[image]

the structure can only hold at max 29 bit pointers. So how would the 32 pointer fit in?

LaszloLango, 2018-06-01:
`ecma_value_t` cannot contain any pointer. `ecma_value_t` and `jmem_cpointer_t` arre two different types.

zherczeg, 2018-06-01:
with cpointer 32 bit you must use 8 byte aligned pointers. jerry allocator satisfy this requirement, and most mallocs as well. Btw the image is outdated, we use 3 bit for type now, and no error bit.

jack9766, 2018-06-01:
The following is pulled directly from the doc:

> in case of number, string and object the value contains an encoded pointer.

Is it outdated or `ecma_value_t` does store pointers in it's value section?

I assume by "encoded" ptr they mean compressed?

zherczeg, 2018-06-01:
Depends. On 64 it contains a 16 or 32 bit cpointer, on 32 bit it is a direct pointer.

jack9766, 2018-06-01:
Will disabling compressed pointer lower performance? since dereferencing uncompressed ptr might take longer?

zherczeg, 2018-06-01:
Hm? I would say the opposite.

jack9766, 2018-06-01:
I tested with the flag on and off. It seem using uncompressed ptr increased the performance of overheads(JS calling C, C to JS, JS to JS) by 5 to 15 percent. But the performance of JS function bodies stay approximately the same.

Any explanations?

closed 2018-06-01 by jack9766