jerryscript-project/jerryscript #2383: Incorrect output from JSON.stringify

renatahodovan, 2018-06-05:
###### Jerry version:

```
Checked revision: 8b8bced6
Build command: ./tools/build.py --clean --debug
```
###### OS:

```
Ubuntu 17.10
```
###### Description:

The test case below converts an array to JSON string. However, (probably because of the unicode characters) the last object's last field (`fz_return`) will be somehow truncated, resulting an incorrect JSON string.

###### Test case:

```javascript
print(JSON.stringify([
	{"dddddd":"ccccccccccccc","bbbbb":225,"vvvvv":{"aaaaaaa":"call","ffffffff":["w2\u1D306A"],"fz_return":0}},
	{"dddddd":"ccccccccccccc","bbbbb":225,"vvvvv":{"aaaaaaa":"call","ffffffff":["\uD834\uDF06"],"fz_return":0}},
	{"vvvvv":{"aaaaaaa":"call","fffffffff":["D"],"fz_return":0}}]))
```

###### Expected output:

```javascript
[{"dddddd":"ccccccccccccc","bbbbb":225,"vvvvv":{"aaaaaaa":"call","ffffffff":["w2ᴰ6A"],"fz_return":0}},{"dddddd":"ccccccccccccc","bbbbb":225,"vvvvv":{"aaaaaaa":"call","ffffffff":["𝌆"],"fz_return":0}},{"vvvvv":{"aaaaaaa":"call","fffffffff":["D"],"fz_return":0}}]
```

###### Actual output:

```javascript
[{"dddddd":"ccccccccccccc","bbbbb":225,"vvvvv":{"aaaaaaa":"call","ffffffff":["w2ᴰ6A"],"fz_return":0}},{"dddddd":"ccccccccccccc","bbbbb":225,"vvvvv":{"aaaaaaa":"call","ffffffff":["������"],"fz_return":0}},{"vvvvv":{"aaaaaaa":"call","fffffffff":["D"],"fz_r:0}}]
```

galpeter, 2018-06-06:
Here's a minimized test case:
```js
print("\uD834\uDF06")
```
and the problem is in the print handler implementation which uses the `jerry_substring_to_char_buffer` method instead of the `jerry_substring_to_utf8_char_buffer` method to get the character sequence to print out.

akosthekiss, 2018-08-03:
I wanted to close this issue as a PR has landed that claimed to fix it but found problem instead.

With @galpeter's minimized test case, I get:
```
$ jerryscript/build/bin/jerry issue-2383bis.js
𝌆
```
This seems to be OK. But with the originally reported test case, I get:
```
$ jerryscript/build/bin/jerry issue-2383.js 
[{"dddddd":"ccccccccccccc","bbbbb":225,"vvvvv":{"aaaaaaa":"call","ffffffff":["w2ᴰ6A"],"fz_return":0}},{"dddddd":"ccccccccccccc","bbbbb":225,"vvvvv":{"aaaaaaa":"call","ffffffff":["𝌆"],"fz_return":0}},{"vvvvv":{"aaaaaaa":"call","fffffffff":["D"],"fz_ret0}}]
```
That is, the unicode character shows up properly but the output is still invalidly truncated (see `"fz_ret0}}]` at the end).

@imiklos, as the author of the related PR, could you please take another look at this issue?

(Validated on latest jerry, 64051b5b.)

imiklos, 2018-08-06:
@akosthekiss Sure, I will take another look.

DanielBallaSZTE, 2018-09-04:
This issue can be closed as #2483 fixes the issue.

closed 2018-09-04 by LaszloLango