jerryscript-project/jerryscript #2393: Linking issue for ARM on OSX

AnthonyCalandra, 2018-06-09:
Hello,

I'm trying to get Jerryscript running on an ARM device but I'm running into a linker problem I can't seem to figure it out. I've added a toolchain file named toolchain_gba_arm7tdmi and it looks like:
```cmake
set(DEVKITARM $ENV{DEVKITARM})
set(DEVKITPRO $ENV{DEVKITPRO})

if(DEVKITPRO)
	message(FATAL_ERROR "DEVKITPRO environment variable not set")
endif()

if(NOT DEVKITARM)
	message(FATAL_ERROR "DEVKITARM environment variable not set")
endif()

set(CMAKE_SYSTEM_NAME GBA)
set(CMAKE_SYSTEM_PROCESSOR arm7)
set(FLAGS_COMMON_ARCH -mthumb -mtune=arm7tdmi -mcpu=arm7tdmi -fomit-frame-pointer -Wno-packed-bitfield-compat -ffast-math)

set(CMAKE_C_COMPILER ${DEVKITARM}/bin/arm-none-eabi-gcc)
set(CMAKE_C_COMPILER_WORKS TRUE)
```
As you can see, it uses the `arm-none-eabi-gcc` compiler. When I build Jerryscript I get no compilation errors, but I've noticed during the linking steps I get these messages:
```
[ 85%] Linking C static library ../lib/libjerry-core.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: ../lib/libjerry-core.a the table of contents is empty (no object file members in the library define global symbols)
[ 85%] Built target jerry-core
```
I'm only taking a guess here, but I think it's not using the ARM linker and instead using Xcode. Before this message however, during the start of the build process, I get a cmake warning:
```
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_TOOLCHAIN_FILE
```

I'm assuming this is why I get undefined reference when I link Jerryscript with my main project. This is how I build Jerryscript:
```
python $(CURDIR)/jerryscript/tools/build.py --profile=es2015-subset --jerry-libc=off \
		--cpointer-32bit=off --mem-heap=256 \
		--toolchain=$(CURDIR)/jerryscript/cmake/toolchain_gba_arm7tdmi.cmake \
		--jerry-cmdline=off
```

Any help would be appreciated! If you need any more info please let me know. Thanks.

akosthekiss, 2018-06-09:
Hi,

It might help diagnosing the issue if you could post a bit more console output, especially the beginning of the build where cmake status messages are printed with the identified C compiler, platform, etc. Increasing the verbosity (`-v`) may also help.

AnthonyCalandra, 2018-06-09:
I've attached the output to this post.
[build.txt](https://github.com/jerryscript-project/jerryscript/files/2087318/build.txt)

akosthekiss, 2019-01-21:
Closing as the issue has been tracked down and solution is available. @AnthonyCalandra feel free to reopen the issue or open a new one if the proposed approach does not work for you.

closed 2019-01-21 by akosthekiss