jerryscript-project/jerryscript #2402: const keyword in JavaScript

BernardXiong, 2018-06-19:
Could JerryScript support const keyword?

for example, 
```javascript
const ctx = wx.createCanvasContext('myCanvas')

// Create linear gradient
const grd = ctx.createLinearGradient(0, 0, 200, 0)
grd.addColorStop(0, 'red')
grd.addColorStop(1, 'white')

// Fill with gradient
ctx.setFillStyle(grd)
ctx.fillRect(10, 10, 150, 80)
ctx.draw()
```

zherczeg, 2018-06-19:
The lexical environment rules for const and let makes it very hard to implement it in an Abstract-Syntax-Tree less environments. At least a somewhat efficient way. Until somebody came with a clever idea how to do this it will probably won't happen.

BernardXiong, 2018-06-20:
or just simply let 'const' as 'var'? Or add a read-only flag in var?

zherczeg, 2018-06-20:
Possible but will work in a different way that people expect.

BernardXiong, 2018-06-20:
Yes. But better than `parse failed`.

How about the specification of 'const' key word in JavaScript?

LaszloLango, 2018-06-20:
I agree with @zherczeg. Non-standard behavior is something we should avoid. People usually expects standard behavior. You can transcript your ES2015 code to ES5.1 with babel before you run JerryScript. It does what you want (changes 'const' to 'var').

BernardXiong, 2018-06-20:
OK. I see. It's the feature of ES2015/ES6. Thank you!

closed 2018-06-20 by BernardXiong