jerryscript-project/jerryscript #2412: How to disable TRACE_MSG ?

tkymeno2, 2018-06-28:
Hi,

I'd like to turn off print message log level trace, however it looks like always be printed on console because the macro JERRY_TRACE_MSG always are to be replaced jerry_port_log method when compiling.
e.g.) ecma-regexp-object.c

On the other hand, I could turn off log level error one by defining JERRY_NDEBUG macro.

Is there anything good way to turn off lower leve log to be printed ?

LaszloLango, 2018-06-29:
@tkymeno2 hi, do you use the default main implementation of JerryScript? In that case you can use the `--log-level NUM` runtime argument to set the log level. If you are using your own main implementation, then it depends on your `jerry_port_log` implementation. See: http://jerryscript.net/port-api/#io-1
See also the related part in `jerryscript-port.h`:
```
/**
 * Jerry log levels. The levels are in severity order
 * where the most serious levels come first.
 */
typedef enum
{
  JERRY_LOG_LEVEL_ERROR,    /**< the engine will terminate after the message is printed */
  JERRY_LOG_LEVEL_WARNING,  /**< a request is aborted, but the engine continues its operation */
  JERRY_LOG_LEVEL_DEBUG,    /**< debug messages from the engine, low volume */
  JERRY_LOG_LEVEL_TRACE     /**< detailed info about engine internals, potentially high volume */
} jerry_log_level_t;
```

tkymeno2, 2018-07-03:
@LaszloLango Thank you for your rapid answering.

I'm using own main implementation on mbed platform.
In this case, I can control logs by editing `jerry_port.c` when I port it, since it says log ignore setting.
Is my understanding loosely right ?

```
/**
 * Provide log message implementation for the engine.
 */
void
jerry_port_log (jerry_log_level_t level, /**< log level */
                const char *format, /**< format string */
                ...)  /**< parameters */
{
  (void) level; /* ignore log level */

  va_list args;
  va_start (args, format);
  vfprintf (stderr, format, args);
  va_end (args);
} /* jerry_port_log */
```

LaszloLango, 2018-07-03:
@tkymeno2 your implementation is ignoring the `level` parameter, so this is why you see all of the messages. If you don't want to do it configurable in runtime, then you can simply write this:

```c
if (level <= JERRY_LOG_LEVEL_ERROR)
{
  va_list args;
  va_start (args, format);
  vfprintf (stderr, format, args);
  va_end (args);
}
```

tkymeno2, 2018-07-03:
@LaszloLango Now I can control log level and disable messages. Thank you 👍

closed 2018-07-03 by tkymeno2