jerryscript-project/jerryscript #2417: jerryscript baremetal

wajahat-ublox, 2018-07-04:
Can i build jerryscript without any OS for a target e.g. stm32f4? I cant seem to find any clear instructions for building it for a baremetal MCU. Also in the targets folder, stm32f4 is with either Nuttx or Riot but no baremetal. Can someone confirm this?

LaszloLango, 2018-07-04:
There is no baremetal support for any certain MCU at this point.

akosthekiss, 2018-07-04:
Actually, JerryScript _should_ work on baremetal, too. Except for its debugger server component, which can be disabled (or, is disabled by default), the project has no OS dependencies. So, if you have a compiler with a decent libc, you should be fine. But, of course, you'll have to write the porting layer (a handful of functions only) and the application that embeds/calls the JerryScript library.

wajahat-ublox, 2018-07-04:
I get the idea however currently such an implementation does not exist is that right?

akosthekiss, 2018-07-04:
That's correct, no such example exists in the code base right now.

closed 2018-07-04 by wajahat-ublox