jerryscript-project/jerryscript #2433: multithreaded jerryscript

daeyeon, 2018-07-23:
I'd like to know your thoughts about an isolated execution model in jerryscript for asynchronous multithreading. What are needed to support the following features of platforms based on jerryscript?

- Isolated instances of jerryscript engine.
- Communication with each threaded-instance (e.g message passing or something like SharedArrayBuffer)
- Globally shared objects protected from concurrent access.

zherczeg, 2018-07-23:
I think the features for this are more-or-less there.

- Isolation threads: if JERRY_ENABLE_EXTERNAL_CONTEXT is enabled, jerry_port_get_current_instance returns the current jerryscript instance.
- It is possible to create ArrayBuffers into areas provided by pointers. Hence multiple JerryScript instances can use the same memory area. Messages can also be transmitted through C callback functions.
- Globally shared objects can be created by C callback functions and accessors. These C functions can ensure the necessary protection. Muliple JerryScript instances cannot share the access to the same object since they don't share their memory allocators.

akosthekiss, 2018-07-23:
cc @galpeter

daeyeon, 2018-07-23:
Regarding globally shared objects, let's say that there are codes for a module, `console`, which could be shared on threaded-instances. Let's think about any immutable data. It would be memory-efficient to provide a way to share them with other instances.

zherczeg, 2018-07-23:
The issue is compressed pointers. Each JerryScript has an own heap, and compressed pointers points to a data inside it. Hence different JerryScript instances cannot reference data in others' heap. Maybe with some mmap magic we could map a single shared region at the end of each private heap, but that is also non-trivial (if Linux allows it).

rerobika, 2019-10-25:
The corresponding https://github.com/jerryscript-project/iotjs/issues/1716 has already been closed. Closing the issue due to inactivity, please reopen it if needed.

closed 2019-10-25 by rerobika