jerryscript-project/jerryscript #2490: Assertion 'ecma_is_value_integer_number (index_val)' failed in ecma_builtin_promise_all_handler

renatahodovan, 2018-08-28:
###### Jerry version:

```
Checked revision: 300e40ba
Build command: tools/build.py --debug --logging=on --profile=es2015-subset
```

###### OS:

```
Ubuntu 18.04, x86_64
```

###### Test case:

```javascript
Object.defineProperty(Object.prototype, 6, {});
Promise.all([0]);
```

###### Backtrace:

```
ICE: Assertion 'ecma_is_value_integer_number (index_val)' failed at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-promise.c(ecma_builtin_promise_all_handler):348.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff7a24801 in __GI_abort () at abort.c:79
#2  0x000055555555f364 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-port/default/default-fatal.c:71
#3  0x000055555558ac1f in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.c:58
#4  0x000055555558ac71 in jerry_assert_fail (assertion=0x5555555da2b8 "ecma_is_value_integer_number (index_val)", 
    file=0x5555555da060 "jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-promise.c", 
    function=0x5555555d3e20 <__func__.5164.lto_priv.262> "ecma_builtin_promise_all_handler", line=348)
    at jerryscript/jerry-core/jrt/jrt-fatals.c:82
#5  0x0000555555587d41 in ecma_builtin_promise_all_handler (function=867, this=72, argv=0x5555558f18dc <jerry_global_heap+1324>, 
    argc=1) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-promise.c:348
#6  0x00005555555a0776 in ecma_op_function_call (func_obj_p=0x5555558f1710 <jerry_global_heap+864>, this_arg_value=72, 
    arguments_list_p=0x5555558f18dc <jerry_global_heap+1324>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.c:589
#7  0x00005555555a1955 in ecma_process_promise_reaction_job (obj_p=0x5555558f18d8 <jerry_global_heap+1320>)
    at jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c:170
#8  0x00005555555a1d73 in ecma_process_all_enqueued_jobs ()
    at jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c:343
#9  0x00005555555befcb in jerry_run_all_enqueued_jobs () at jerryscript/jerry-core/api/jerry.c:570
#10 0x00005555555bdcf1 in main (argc=3, argv=0x7fffffffdc58) at jerryscript/jerry-main/main-unix.c:863
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

renatahodovan, 2018-08-28:
Changing the index (`6`) in the `Object.defineProperty` call results further assertion failures:

###### Changing to `2`:
```ICE: Assertion 'ecma_is_value_object (capability) && ecma_is_value_object (array) && ecma_is_value_object (ctor)' failed at /home/reni/work/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-promise.c(ecma_builtin_promise_do_all):397.```
Although, the failure happens in different method, but it's similar to #2468 

###### Changing to `3`:
```ICE: Assertion 'ecma_is_value_object (value)' failed at /home/reni/work/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c(ecma_get_object_from_value):659.```
This might be related to #2465 .

###### Changing to `4`:
```ICE: Assertion 'ecma_is_value_boolean (handler) || ecma_op_is_callable (handler)' failed at /home/reni/work/jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c(ecma_process_promise_reaction_job):158.```

###### Changing to `5`:
```ICE: Assertion 'ecma_is_value_boolean (already_called)' failed at /home/reni/work/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-promise.c(ecma_builtin_promise_all_handler):322.```

###### Changing to `7`:
```ICE: Assertion 'ecma_is_value_object (value)' failed at /home/reni/work/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c(ecma_get_object_from_value):659.```
This might be related to #2465 

###### Changing to `8`:
```ICE: Assertion 'ecma_is_value_object (remaining)' failed at /home/reni/work/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-promise.c(ecma_builtin_promise_remaining_inc_or_dec):274.```

DanielBallaSZTE, 2018-08-29:
Number 2 gets fixed with #2491

PR #2526 (2018-09-24): Change Promise properties to internal properties
There was an issue in the Promise implementation where properties were accessible from JavaScript.
ie. `Object.defineProperty(Object.prototype, 0, {})` could modify properties which should've been inaccessible. That sets the `writable` value to `false`, therefore when creating a Promise object, creating their internal properties with `ecma_op_object_put` failed. The reason behind that is somewhat interesting as 0-7 were the same values as the enum values in the property list of the Promise object.
Changing these properties to internal, makes the [... cut]
commit 7717d2ee (2018-09-24): Change Promise properties to internal properties (#2526)

There was an issue in the Promise implementation where properties were accessible from JavaScript.
ie. `Object.defineProperty(Object.prototype, 0, {})` could modify properties which should've been inaccessible.
The reason behind that is somewhat interesting as 0-7 were the same values as the enum values in the property list of the Promise object.
Changing these properties to internal, makes them inaccessible from JS side.

Also some tests have been changed, namely 2490 and 2465.
The 2490 one got renamed, and all of the testcases from th [... cut]
closed 2018-09-24 by LaszloLango