jerryscript-project/jerryscript #2539: ArrayBuffer tutorial/help, please...

t-harvey, 2018-09-21:
Folks,
I apologize if I'm being obtuse, but I'm stuck:

This morning, I cloned a clean repo and rebuilt a unix version of the repl (using jerry-main/main-unix.c).

The following Javascript:
`var x = new ArrayBuffer(128);`

...produces:
`Script Error: ReferenceError`

...I've also tried to use the API to create my own ArrayBuffer.  In the following code, I've created two objects, one a long value (as a control) and one an ArrayBuffer -- the code is the same for both, with the exception of the creation of the actual data object -- and I've inserted the code into the main routine before the repl starts:

   ```
   jerry_value_t new_thing = jerry_create_string((jerry_char_t *)"new_thing");
   jerry_value_t new_thing2 = jerry_create_string((jerry_char_t *)"new_thing2");
   
   jerry_value_t new_ab = jerry_create_arraybuffer(10);
   jerry_value_t new_num = jerry_create_number(4.0);

   jerry_value_t global_object = jerry_get_global_object();

   jerry_set_property(global_object, new_thing, new_ab);
   jerry_set_property(global_object, new_thing2, new_num);

   jerry_release_value(global_object);
```

...and inside the repl:
```
jerry> new_thing
Script Error: ReferenceError
jerry> new_thing2
2
```

...so what am I doing wrong?

Thanks!
Tim

galpeter, 2018-09-21:
Hi,

by default the ArrayBuffer is turned off (and also other post es5.1 features). To enable it there is a profile system in place (https://github.com/jerryscript-project/jerryscript/tree/master/jerry-core/profiles).
For a quick test, you can use the following cmd:
```
$ ./tools/build.py --profile=es2015-subset
```
With this you should have ArrayBuffer in the repl.

t-harvey, 2018-09-21:
Peter,
Perfect -- that did it.

Thanks!
Tim

closed 2018-09-21 by t-harvey