jerryscript-project/jerryscript #2552: jerry_get_property_by_index returns NULL when array item initialized with zero

imorrisonca, 2018-10-04:
Hi,

We are seeing some behavior which we didn't expect when processing an object (jerry_value_t) in c, returned from a javascript function.  I am not sure if this is normal behavior, as I am not very familiar with Javascript.  Any comments or responses to shed some light on this would be appreciated.

We parse and run the following script:
`function(input) {
    var x = 0;
    var a = [x];
    return { "name_x" : a };
}";`

Then, in c, we extract the array `a` and check it's type and size using: `jerry_value_is_array(a)` and `jerry_get_array_length(a)`.  These return `true` and `1`, as expected.

We then call `jerry_get_property_by_index(a, 0)`.  This call returns NULL, which is unexpected.  We were expecting the call to return an object containing a float value of 0.

Is this normal?

akosthekiss, 2018-10-05:
There has been no proper minimal reproduction case shown, so it's hard to tell what's wrong. I've written a unit-test-like file and placed it `tests/unit-core/test-issue-2552.c`:

```c
#include "jerryscript.h"
#include "test-common.h"

int
main (void)
{
  TEST_INIT ();

  jerry_init (JERRY_INIT_EMPTY);

  const jerry_char_t js[] = "function f(input) { var x = 42; var a = [x]; return { 'name_x' : a }; }; f().name_x";
  jerry_value_t a = jerry_eval (js, sizeof (js) - 1, JERRY_PARSE_NO_OPTS);

  TEST_ASSERT (jerry_value_is_array (a));
  TEST_ASSERT (jerry_get_array_length (a) == 1);

  jerry_value_t a0 = jerry_get_property_by_index (a, 0);

  TEST_ASSERT (jerry_value_is_number (a0));
  TEST_ASSERT (jerry_get_number_value (a0) == 42);

  jerry_release_value (a0);
  jerry_release_value (a);

  jerry_cleanup ();

  return 0;
} /* main */
```

(I've slightly modified the values from those reported as 0 might potentially come from casting different values, e.g., `(int) false`, `(int) NULL`, so it'd might be hard to tell why we are seeing a zro somewhere. 42 is safer.)

When running the so-extended unit tests, I get "PASS: build/tests/unittests-es5.1-debug/tests/unit-test-issue-2552". So, please, give some more details.

imorrisonca, 2018-10-05:
Hello, and thank you for the quick response.

The unit-test file: test-issue-2552.c above, will behave in the same way as our code if `var x = 42;` is changed to `var x = 0;`.  This really is the origin of my question.

In our application, the array `a` is being populated with numeric data, which we can not control.  Sometimes, the value of that data is zero.  For example:  [ 1.5, 23.3, 0, 17 ].  We then attempt to access that data from c code.  The data `0` is valid to us but the parser appears to be interpreting the `0` as a null element.  Is this the correct behavior, according to the ECMA spec.?  I wasn't able to determine this myself.

Thank you

akosthekiss, 2018-10-05:
I'm not sure what "null element" could mean in this context. When calling `jerry_get_number_value`, it returns a `double` no matter what. That cannot be null (pointer?), only 0.0. Based on your comment, I've created a variant of the above test case `tests/unit-core/test-issue-2552bis.c`:

```c
#include "jerryscript.h"
#include "test-common.h"
#include <stdio.h>

int
main (void)
{
  TEST_INIT ();

  jerry_init (JERRY_INIT_EMPTY);

  const jerry_char_t js[] = "function f() { var a = [ 1.5, 23.3, 0, 17 ]; return { 'name_x' : a }; }; f().name_x";
  jerry_value_t a = jerry_eval (js, sizeof (js) - 1, JERRY_PARSE_NO_OPTS);

  TEST_ASSERT (jerry_value_is_array (a));
  TEST_ASSERT (jerry_get_array_length (a) == 4);

  jerry_value_t a0 = jerry_get_property_by_index (a, 0);
  TEST_ASSERT (jerry_value_is_number (a0));
  TEST_ASSERT (jerry_get_number_value (a0) == 1.5);
  jerry_release_value (a0);

  jerry_value_t a1 = jerry_get_property_by_index (a, 1);
  TEST_ASSERT (jerry_value_is_number (a1));
  TEST_ASSERT (jerry_get_number_value (a1) == 23.3);
  jerry_release_value (a1);

  jerry_value_t a2 = jerry_get_property_by_index (a, 2);
  TEST_ASSERT (jerry_value_is_number (a2));
  TEST_ASSERT (jerry_get_number_value (a2) == 0.0);
  jerry_release_value (a2);

  jerry_value_t a3 = jerry_get_property_by_index (a, 3);
  TEST_ASSERT (jerry_value_is_number (a3));
  TEST_ASSERT (jerry_get_number_value (a3) == 17.0);
  jerry_release_value (a3);

  jerry_release_value (a);

  jerry_cleanup ();

  return 0;
} /* main */
```

This test case also runs fine, "PASS: build/tests/unittests-es5.1-debug/tests/unit-test-issue-2552bis". Which part of this code does not work as you'd expect?

akosthekiss, 2018-10-09:
@imorrisonca Have you had a chance to look at the latest example snippet (above)? Does the issue still exist?

imorrisonca, 2018-10-09:
Hi @akosthekiss ,

Sorry for the delay.  I have tested again and found my issue.  The problem was that I assumed that the following represents a NULL:  `(jerry_value_t)0;`, which it does not.  I was testing this way:  
`
jerry_value_t b = jerry_get_property_by_index(a, index);
if  (!b) { // error here
`
I see now that this test will pass:  `TEST_ASSERT (jerry_value_is_number ((jerry_value_t)0));`.  I have changed my code to use:  `jerry_value_is_null(b)` and it works fine now.

Thank you very much for your help

closed 2018-10-09 by imorrisonca