jerryscript-project/jerryscript #2562: Scrap the jerry_object_native_info_t type and just use jerry_object_native_free_callback_t for native objects

t-harvey, 2018-10-17:
Folks,
Disclaimer: if I've misunderstood something in my explanation: a) mea culpa; feel free to beat me up :-), and b) please explain the truth not only to me, but maybe also include the explanation in the docs...

The jerry_object_native_info_t structure contains exactly one field, a pointer to the function that the garbage collector calls to free the native object.  A crude form of type checking is performed every time the native object is accessed: the address of the native_info_t structure is recorded when the native object is created, and subsequent accesses include a check against this address.

...this design has bitten me, but instead of trying to explain the vagaries of separate compilation and const pointers (I'll gladly do so, but it'll be long and tedious... :-), I'd rather make the case to change this from a design perspective:

In general, a structure with only one field does not need to exist -- the field can just be used as it is; as a simple example: imagine if every integer were a struct containing a single int...

Of course, some systems do exactly that: wrap a struct around simple data types, but they do so for really specific reasons, none of which(?) apply to the native-object code: the "type checking" is only against a constant address of a structure -- probing the system indicates that the contents of the structure aren't even accessed; but even if they are, the only thing in the structure is a function pointer that, by definition, is itself a constant address.

Can we fix this so that the native-object code just makes sure that the free_cb function remains unchanged?  (That is, wherever we now send in a pointer to the native_info_t struct, instead send in the free_cb-function pointer instead.)

(No need to read this paragraph is you're already convinced :-) -- The current abstraction is excellent, in that it says: every time you access this otherwise-opaque pointer, make sure it's the right kind of thing by checking against a type signature (which happens to just be the pointer used to free the object).  If we implement my proposed change, the abstraction becomes: every time you access this pointer, make sure the guy accessing it knows that a particular function is the function used to free it -- yuck.  So to maintain the abstraction with the native_object_t struct, we should not check the address of the struct, but, rather, do a field-by-field comparison of the structs.  That way, two otherwise identical structs would match.  This, obviously, would require more processing, but would maintain the nice design.

Thanks,
Tim

P.S. This is not a particularly urgent issue -- careful #def'ing solved my specific problem, but it was a little tricky to get right, and the docs aren't very helpful -- thus this suggestion to simplify the design.

LaszloLango, 2018-10-31:
CC: @zherczeg

zherczeg, 2018-11-06:
I may misunderstand something but JerryScript internally does not do any "form of type checking". From JerryScript perspective, these are just pointers. The user application needs to do its type checks on their own way.

Historically we had a different user and callback pointers, but only the user pointers could be retrieved. Therefore no type checks was available. Later the current approach was added, and we had to maintain two approaches. It was not convenient, and I wouldn't want to do it again.

The motivation of using a structure pointer was that the structure can have other members (mostly type info), and you can use this for sub-classing objects. Your C code can check the callback, and if it matches, you can type cast the pointer to another structure.  If we pass only the callback, this is not possible anymore, which sounds a step back.

rerobika, 2019-10-18:
There is no activity on this issue for a year and the question seems to be answered for me. I'm closing the issue for now, please feel free to reopen it if needed.

closed 2019-10-18 by rerobika