jerryscript-project/jerryscript #2568: How to use the jerryx_arg* for arrays?

t-harvey, 2018-10-19:
Folks,
I'm stymied, and I'd appreciate a tutorial...

I'm trying to use the jerryx_arg* functions to do parameter checking --the example for jerryx_arg_array shows how to get a static array (e.g., if you expected a parameter that looked like: [int, float, string]), but I'm looking to validate a parameter that is an arbitrarily long array of \<Type>s (e.g., in C, an int[]).

I'm writing my own jerryx_arg_transform_funct_t: the first thing I'll do is pop the element off the args list, which returns a jerry_value_t, but my mental model is that it should instead return an array of jerryx_arg_t's which I'll then need to call jerryx_arg_\<Type> over to build up my array.

...and there's where I'm stuck.  Help is most(!) appreciated!
Tim

P.S. I've already written one transform_func_t for floats, as I needed to encapsulate the type conversion from double (which is provided by jerryx_arg_number, which requires a double*), which I've shown below; any comments would be likewise appreciated:

```
#define interpreter_arg_float(x, z) \
            jerryx_arg_custom(x, \
                             (uintptr_t)&((uintptr_t []){(uintptr_t)JERRYX_ARG_COERCE, \
                                                         (uintptr_t)z}), \
                              interpreter_arg_to_float)

// the transform function for floating-point numbers is defined only
//   for doubles, so we need a wrapper that will call the version for
//   doubles and then convert that value to a float
jerry_value_t interpreter_arg_to_float(jerryx_arg_js_iterator_t *js_arg_iter_p,
                                       const jerryx_arg_t *c_arg_p)
{
    double intermediate_result = 0.0;
    uintptr_t *extra_info = (uintptr_t*)(c_arg_p->extra_info);

    jerryx_arg_t get_double = interpreter_arg_number(&intermediate_result,
                                                     extra_info[0],
                                                     extra_info[1]);
    jerry_value_t convert_to_double_result =
                     get_double.func(js_arg_iter_p, &get_double);

    if (!jerry_value_is_error(convert_to_double_result))
        *((float*)(c_arg_p->dest)) = (float)intermediate_result;
    return convert_to_double_result;
} /* interpreter_arg_to_float */
```

t-harvey, 2018-10-24:
Folks,
Stand down...  I know that the entire community was scrambling desperately to solve this (minor) brain teaser, but, necessity being a mother, I've hit upon a working/workable solution that I'd like to submit for comments (and, hopefully, this thread will serve as a tutorial for future jerryx_arg users) :

The trick to writing a custom jerryx_arg function is understanding that at some point you're going to have a perfectly good jerry_value_t, despite the fact that the focus is on writing a function that returns a jerryx_arg_t.  (This is what messed me up; I assumed that the parameters were being stored in some undefined data structure that could only be understood by the jerryx_arg_* functions.)  ("No duh!", I hear the experts cry -- but the whole point of a (well described) API is to keep the users from digging through actual code... :-)

So I've got a parameter that wants an array of arrays of C integers, and here is the code I've written.

First, the jerryx_arg_transform_func_t (I'm using "interpreter" in my own names rather than "jerryx" to avoid confusion, but the naming, call signatures, etc. otherwise follow the pattern of jerryx_arg functions).  This is the outward-facing function that users utilize; it returns a jerryx_arg_t struct that can be used for other jerryx_arg calls.

```
/* "x" = pointer to an array of arrays, z = JERRYX_ARG_OPTIONAL or JERRYX_ARG_REQUIRED */
/* CAUTION: calls malloc */
#define interpreter_arg_int32_t_array_array(x, z) \
            jerryx_arg_custom(x, \
                              (uintptr_t)&((uintptr_t []){(uintptr_t)JERRYX_ARG_COERCE, \
                                                         (uintptr_t)z}), \
                              interpreter_arg_to_int32_t_array_array)
```
...the transform function of which is:
```
jerry_value_t interpreter_arg_to_int32_t_array_array(jerryx_arg_js_iterator_t *js_arg_iter_p,
                                                     const jerryx_arg_t *c_arg_p)
{
    /* grab the top of the arg stack and do our own testing on it */
    jerry_value_t array_value = jerryx_arg_js_iterator_pop(js_arg_iter_p);
    if (!jerry_value_is_array(array_value))
        return jerry_create_error(JERRY_ERROR_TYPE,
                                  (const jerry_char_t *) "Expected array.");

    /* NOTE: our "*_array" types are structs containing two fields: "length", the number
       of elements in the array, and "items", the array of values */
    unsigned int array_length = jerry_get_array_length(array_value);
    int32_t_array_array *new_int32_t_array_array = malloc(sizeof(int32_t_array_array));

    new_int32_t_array_array->length = array_length;
    if (array_length > 0)
        new_int32_t_array_array->items = (int32_t_array *)malloc(array_length*sizeof(int32_t_array));

    /* now, use other jerryx_arg functions on each element of the array we're holding */
    jerry_value_t get_element_return_value;
    for (int i = 0; i < array_length; i++)
    {
        jerry_value_t input_mapping[1] = { jerry_get_property_by_index(array_value, i) };
        /* note: we know the contents of "extra_info", b/c we built the array in the #def, above... */
        jerryx_arg_t output_mapping[1] = {
                       interpreter_arg_int32_t_array(&(new_int32_t_array_array->items[i]),
                                                     ((uintptr_t*)(c_arg_p->extra_info))[1])
                                          };
        get_element_return_value =  jerryx_arg_transform_args(input_mapping, 1,
                                                              output_mapping, 1);
        if (jerry_value_is_error(get_element_return_value))
        {
            /* TODO: free (ALL!) memory!!! */
            return get_element_return_value;
        }
    }

    /* fallthrough indicates that the array was read in correctly */
    *((int32_t_array_array *)(c_arg_p->dest)) = *new_int32_t_array_array;
    return get_element_return_value;
} /* interpreter_arg_to_int32_t_array_array */
```

...Hopefully, that's clear enough (drop me a line if not), but we'll show the code to get a single array of integers (which is what the above code calls) to make obvious the pattern:

```
#define interpreter_arg_int32_t_array(x, z) \
            jerryx_arg_custom(x, \
                             (uintptr_t)&((uintptr_t []){(uintptr_t)JERRYX_ARG_COERCE, \
                                                         (uintptr_t)z}), \
[... 24 lines cut ...]
                                               ((uintptr_t*)(c_arg_p->extra_info))[1])
                                          };
        get_element_return_value = jerryx_arg_transform_args(input_mapping, 1,
                                                             output_mapping, 1);
        if (jerry_value_is_error(get_element_return_value))
        {
            /* TODO: free (ALL!) memory!!! */
            return get_element_return_value;
        }
    }

    /* fallthrough indicates that the array was read in correctly */
    *((int32_t_array *)(c_arg_p->dest)) = *new_int32_t_array;
    return get_element_return_value;
} /* interpreter_arg_to_int32_t_array */

/* we have defined all of our "interpreter_arg_*" functions to have the same signature and 
   have "#define" wrappers for the supplied jerryx_arg functions -- so "ROUNDING_POLICY"
   is of type jerry_x_round_t, "CLAMPING_POLICY" is of type jerryx_arg_clamp_t, and
   "COERCION_POLICY" is of type jerryx_arg_coerce_t */
#define interpreter_arg_int32_t(x, z)   jerryx_arg_int32(x, ROUNDING_POLICY, \
                                                          CLAMPING_POLICY, \
                                                          COERCION_POLICY, \
                                                          (jerryx_arg_optional_t)z)
```

LaszloLango, 2018-10-31:
Hi Tim, could you solve this? What is the current status?
You can find the related documentation here: http://jerryscript.net/ext-reference-arg/#jerryx_arg_array
Any feedback on the current documentation is welcome. If you would like to contribute then feel free to open a PR with doc. updates.

t-harvey, 2018-10-31:
LL,
If I understand your question, you’re proposing that I use jerryx_arg_array to
copy my array of elements, right?

The problem with jerryx_arg_array, as I read it, is that it is designed to load,
essentially, an object that is in array form: that is, the user might have sent
in an array with a specific number (and type) of elements, and the handler
has to already know that structure.  (In the documentation, the example shows
the handler loading up a mapping of 1 boolean and 2 doubles and single
instances of those variables are provided to write into.)

What I was trying to do is load up an arbitrary length array (i.e., I don’t know
the length, although I do know the kinds of elements).

The way I read it, using jerryx_arg_array says, “I’ve got an array that holds
a set of things”, whereas what I am trying to ask for is, “I’ve got an array
of unknown length of things, and I need to iterate over the array and copy
it to my local array.”

This is more complicated than we can probably do in a library call, if
only b/c I need to malloc the array after I find out what its length is.
This is similar to the string problem – the library provides a “get_string”
function, but it requires the user to pass in an already allocated buffer,
which we then have to make sure we don’t overwrite, we have to report
an error if it’s not big enough, etc. – the provided solution for strings
is just fine, and we could provide a similar one for arrays (the current
implementation of jerryx_arg_array is sort of doing this), but
jerryx_arg_array is not really a general solution right now.

(In case that wasn’t clear: imagine if jerryx_arg_string worked the same
way as jerryx_arg_array: you’d have to load up a mapping structure that
looked like:
char my_string[32];
const jerryx_arg_t string_mapping[] = {
    jerryx_arg_char(&my_string[0], ...),
    jerryx_arg_char(&my_string[1], ...),
         ...30 more times...
};
/* ...and then you’d set up the array with: jerryx_arg_array(&string_mapping, 32, ...); */

...and what if the string only had 10 characters, not 32?

Instead, jerryx_arg_string says, “give me a buffer, and I’ll fill it with as many
things as there are, up to the buffer’s limit”.  As I say, this might be how
a new jerryx_arg_array would work, but even then, it’s not always going
to be what’s needed.)

Tim

rerobika, 2020-07-29:
Closing the issue due to inactivity.

closed 2020-07-29 by rerobika