jerryscript-project/jerryscript #2581: Add a new enum type to jerryx_arg_* routines

t-harvey, 2018-10-30:
Folks,
tl;dr: I need to "un-pop" the argument stack -- go to the last paragraph (starting with "Finally,...") for my suggested addition to the `jerryx_arg` routines...

I'm working extensively with the `jerryx_arg` routines/types, and I've hit a snag:

I need to query the argument stack without changing it.  For example, I might be expecting a parameter that is _either_ a number or a string.

Thus, I'd like to do something like (simplified, not accurate :-) :
```
result = jerryx_arg_number(...)
if (jerry_value_is_error(result))
    result = jerryx_arg_string(...)
```
I could write my own "`jerryx_arg_number_or_string`", but the permutations of all the arguments are too numerous, it'd be a copy of a whole lot of code that already exists, and I don't get the continuity of behavior with my own code -- specifically, it's not easy to mimic the behavior of the `JERRYX_ARG_COERCE` flag, so a simple call to jerryx_arg_number vs my own homegrown version of the above code would likely give different results given certain inputs.

One solution would be to give more power to the `JERRYX_ARG_OPTIONAL` flag -- if a type check fails, that `jerryx_arg_*` function would not pop the stack.  I dislike that solution, b/c it would mean overloading the meaning of that flag.

A second solution would be to add a flag to all of the `jerryx_arg_*` functions to specify whether the stack top should be popped on failure.  We could add a new enum type like jerry_arg_coerce_t, _etc_., to control this behavior.  I think this solution has the some merit: it keeps the minimalistic interface of the `jerryx_arg_*` functions and lets us maintain the current behavior as the default.  Its downside is that it adds some logic to the `jerryx_arg_*` routines...

To implement this solution, I propose adding a new enum type in the extension code, the documentation for which would read as follows:

> **jerryx_arg_pop_on_failure_t**
> Enum that indicates whether a failed transformation attempt modifies the stack.
> - `JERRYX_ARG_POP_ON_FAILURE` -- calling the transform has the effect of popping the top value off of the argument stack
> - `JERRYX_ARG_DO_NOT_POP_ON_FAILURE` -- if the transform returns an error, the stack is left unchanged

Finally, the solution I'd like to propose is to add a new call to the custom-validation support routines that would let us "restore" the value just popped.  This solution would normally not work, but in our case, "pop" just moves a pointer through an array, so like we increment the pointer on a pop, we decrement that pointer on a restore.  I have written this code (and use it now in my work), and it is as follows:
```
/**
 * Restore the previous JS argument from the iterator.
 * It will change the index and js_arg_p value in the iterator.
 *
 * @return the restored (now current) JS argument.
 */
jerry_value_t
jerryx_arg_js_iterator_restore (jerryx_arg_js_iterator_t *js_arg_iter_p) /**< the JS arg iterator */
{
  if (js_arg_iter_p->js_arg_idx == 0)
    return jerry_create_undefined();
  else
  {
    --js_arg_iter_p->js_arg_idx;
    --js_arg_iter_p->js_arg_p;
    
    return *js_arg_iter_p->js_arg_p;
  }
} /* jerryx_arg_js_iterator_restore */
```

Discussion?
Thanks,
Tim

LaszloLango, 2018-10-31:
Hi Tim,
first of all sorry for the late responses on your recently opened issues. I would change `jerryx_arg_optional_t` to `jerryx_arg_flags_t`. E.g.:
```c
enum {
  JERRYX_ARG_OPTIONAL = (1 << 0),
  JERRYX_ARG_REQUIRED = ~JERRYX_ARG_OPTIONAL,
  JERRYX_ARG_DO_NOT_POP_ON_FAILURE =  (1 << 1),
  JERRYX_ARG_POP_ON_FAILURE = ~JERRYX_ARG_DO_NOT_POP_ON_FAILURE
} jerryx_arg_flags_t;
```

t-harvey, 2018-10-31:
Lazlo,
Thanks!  And your response is only a day (not even) later; the one
I really(!) needed was how to do the arrays (an earlier post), but
I slogged through to a solution, so all’s good.  (If you have a chance
and want to comment, I’d appreciate any advice on that code, too.)

(I’ve never seen your trick with the enum values before (xor’ing the
false values to double the number of flags created by 1<<x) – thanks;
I like learning new techniques!) (I’m old, and I still learn new stuff...)

The problem I have with the enum solution is that it requires us to go in
and rewrite all of the jerryx_arg_* functions to handle the new enums.
The logic there is complicated enough (and spread across multiple
routines) that it doesn’t seem like the best solution.  Also, mixing flags
with different meanings sounds like a bad design: for example, here,
what if it’s an optional parameter but I don’t want to pop the stack
on failure?  (Since I can only specify one of those two conditions, I
can’t get the behavior I want...)

I know it’s a little weird (read: kludgey :-) to “restore” the stack, but
it fits nicely with the way the stack works (we’re already cheating by
just moving an array pointer! :-).  We could allow the user to “push”
the value back onto the stack – the method would be: grab the
top of the stack with peek, call the jerryx_arg_<> routine, and then
push the grabbed value back onto the stack.  This would look better,
but it wouldn’t really be different than the “restore” option.

Tim

zherczeg, 2019-01-16:
Patch landed. Please feel free to reopen it if needed.

closed 2019-01-16 by zherczeg