jerryscript-project/jerryscript #2582: A possible unsigned integer wrap?

360CodeSafe, 2018-10-31:
Hi,
This is Qihoo360 CodeSafe Team, we found a suspicious unsigned integer wrap issue, see
https://github.com/jerryscript-project/jerryscript/blob/3c1d0e61002c5424d2b882a8044b75a11c0dd58c/jerry-core/ecma/base/ecma-helpers-number.c#L539. At line 539, when the program execute `fraction--`, there is a chance that the value of `fraction` is 0. For the condition expression at line 531, if `fraction` and `biased_exp` are all zero, else branch will be chosen.

Since I not familiar with `jerryscript`, I am not sure `fraction` and `biased_exp` are absolutely independent, that is can them be zero at the same time?

Cheers
Qihoo360 CodeSafe Team

LaszloLango, 2018-10-31:
> there is a chance that the value of fraction is 0. 

Do you have a testcase for this? I think this cannot be triggered from an API function.

360CodeSafe, 2018-11-02:
@LaszloLango thank you for your quick response! Unfortunately, there is no test on hand that can trigger this issue given that the issue was found by the static analyzer.

Hmm, this issue may be a false positive if there is no chance that the value of fraction take 0 at line 539.

closed 2018-11-12 by 360CodeSafe