jerryscript-project/jerryscript #2675: global-buffer-overflow in print_unhandled_exception

l0kihardt, 2019-01-06:
# jerry version
aae50c9da089974b2da9bb3df71e0d0dffce467b
# build command
```
python ./tools/build.py --clean --debug --compile-flag="-O1 -g -fsanitize=address -fno-omit-frame-pointer" --lto=off --error-message=on
```
# OS
Ubuntu 18.04
# vulnerability
global buffer overflow
# test case
https://transfer.sh/9BXc6/test.js
# analysis
The global buffer overflow happened in the function `print_unhandled_exception`, here it tries to seek the correct position to print. However the size of `buffer` is limited to 1048576.
If we can make the `pos` over 1048576, the buffer overflow will be triggered.
```C
/* 2. seek and print */
        while (buffer[pos] != '\0')
        {
          if (buffer[pos] == '\n')
          {
            curr_line++;
          }

          if (err_line < SYNTAX_ERROR_CONTEXT_SIZE
              || (err_line >= curr_line
                  && (err_line - curr_line) <= SYNTAX_ERROR_CONTEXT_SIZE))
          {
            /* context must be printed */
            is_printing_context = true;
          }

          if (curr_line > err_line)
          {
            break;
          }

          if (is_printing_context)
          {
            jerry_port_log (JERRY_LOG_LEVEL_ERROR, "%c", buffer[pos]);
          }

          pos++;
        }
```

akosthekiss, 2019-01-06:
Notes to analysis:
- The test case is cca. 2.5 MB, while jerry-main can only deal with cca. 1 MB of input. (For now.)
- `print_unhandled_exception` is a function of jerry-main, i.e., it is not part of the engine but of the command line tool only, which is not really intended for production use. (Unless someone comes up with a real use case with it.)

That doesn't mean that jerry-main couldn't / shouldn't be improved, but it is not that critical.

akosthekiss, 2019-01-17:
@l0kihardt Thanks for the report. I've opened a PR (#2710) to fix this buffer overflow. Could you confirm that it fixes the issue for you?

PR #2710 (2019-01-21): Fix buffer overflow while printing unhandled exception
Fixes #2675

JerryScript-DCO-1.0-Signed-off-by: Akos Kiss akiss@inf.u-szeged.hu
closed 2019-01-21 by robertsipka