jerryscript-project/jerryscript #2677: Diskussion about the possibility of Heap-Spray in JerryScript:

OverDriveGain, 2019-01-07:
Hello everyone, 

Heap Spray is a widely used payload-delivery technique, most JS engines nowadays integrate special mechanisms to mitigate it. Such as nozzle, bubble, etc. I recently came to conclusion that in JerryScript there are also a kind of mitigation to Heap-Spray, can you guys maybe confirm this and tell me what are the mechanisms being used in JerryScript to mitigate Heap-Spray? 

If you refer me to any documentation that I can read considering this issue, I will also be very thankful. 

Thank you so much.

akosthekiss, 2019-01-07:
>  I recently came to conclusion that in JerryScript there are also a kind of mitigation to Heap-Spray

Could you please elaborate a bit on what has lead you to this conclusion?

OverDriveGain, 2019-01-07:
I tried Heap-Spraying in JerryScript. I have analyzed the memory of the heap of JerryScript in run-time. After inserting hexadecimal representation of the Shell Code NOP (which is 0x90), I noticed that JerryScript has automatically added the value 0xc2 after each byte. 
In other words I ran the following script:
var example = unescape("%90%90%90")
And the memory ended up looking like this:
0x90 0xc2 0x90 0xc2 0x90 0xc2

akosthekiss, 2019-01-07:
What you are seeing in the memory is the UTF-8/CESU-8 encoding of U+0090.

OverDriveGain, 2019-01-07:
UTF-8 And CESU-8 are different. So which encoding do you mean exactly?
Unfortunately I couldn't find any resources where it says that the encoding of U+0090 is  (0x90 0xc2) neither in UTF-8 nor in  CESU-8. 

I am writing an encoded value in memory, so it should write it as it is, specially that I am using the unescape() function. And if your statement was true, then the following program:
 var example = unescape("%11%11%11")
shouldn't produce this memory layout:
 0x11 0x11 0x11 

I thank you so much for your collaboration, and I would gladly take your reply as a confirmation of my conclusion.

OverDriveGain, 2019-01-08:
Oh OK. Thank you very much again for your kind answer, it was very helpful. 
You replies were also informative about the UNI code representation. I appreciate it.
Kind regards

closed 2019-01-08 by OverDriveGain