jerryscript-project/jerryscript #2730: is the jerryscript implementation "compileable" or "interpreterable"?

13824125580, 2019-01-25:
take lua for example ,it is obvioulsy compile language because the source language first comiple by luac to mid-binrary code and then execute the byte code by lua vm.
so, is the same logic for jerryscript, i have not found the jerry cunterpart like luac compiler
.thank you.

akosthekiss, 2019-01-25:
Traditionally, JavaScript is an interpreted language because of its highly dynamic nature. However, interpreters can be implemented in various ways: e.g., they may build abstract syntax trees and interpret them by walking these trees; they may compile the sources to some byte code and run a VM executing it; or they may compile the sources to in-memory native code using just-in-time compilation techniques. Whichever way an interpreter is implemented, it may (or may not) choose to serialize its internal representation to disk for later re-use, usually for performance reasons (but reducing memory usage may also be a target). Still, I'd consider all these approaches as interpreters and the language as an interpreted one.

As for lua, I'm not sure about classifying it as an "obviously" compiled language. I'm no lua expert, but AFAIK one doesn't _need_ to compile scripts using `luac` first. The VM can take scripts in  source form and run them (doing the source to byte code compilation at run-time).

The same holds for JerryScript. The JerryScript engine has a parser+compiler part, which translates JS sources to an internal byte code format. By default, the byte code is written into memory buffers, from where the VM component of the engine can execute it. However, the byte code can be written out to files and read back at a different point of time, saving the need for repeated source code parsing, syntax checking, etc. See documentation of internals in the `docs/` folder: https://github.com/pando-project/jerryscript/blob/master/docs/04.INTERNALS.md .

JerryScript calls the saved byte code (+ the accompanying info) snapshots. As it's not the most often used feature of the engine, their support is disabled by default, but the _engine_ can be configured to support them using the `--snapshot-save=on` and `--snapshot-exec=on`  options of the `tools/build.py` script (the first option enabling the saving, the second option enabling the loading+execution of byte code snapshots). Those who are not using the python build wrapper but cmake directly can pass `-DFEATURE_SNAPSHOT_SAVE=ON` and `-DFEATURE_SNAPSHOT_EXEC=ON` to cmake.

On the _command line tool_ side, snapshot support can/should also be enabled by passing `--jerry-cmdline-snapshot=on` to `tools/build.py` (or `-DJERRY_CMDLINE_SNAPSHOT=ON` to cmake). This will build a `jerry-snapshot` tool along the `jerry` command line tool, and it can be used to generate snapshots from script sources. The `jerry` tool can load these snapshots using its `--exec-snapshot FILE` command line option. (See the output of both tools with `--help` for more details.)

I hope this helps.

13824125580, 2019-01-25:
thank you very much

closed 2019-01-25 by akosthekiss