jerryscript-project/jerryscript #2732: Handle scope support in JerryScript

legendecas, 2019-01-28:
Handle scopes greatly simplify variables lifetime management. Also handle scope apis are part of N-API. Would like to have handle scope support in JerryScript natively.

There is a naïve implementation available in ShadowNode project for JerryScript: https://github.com/yodaos-project/ShadowNode/tree/master/deps/jerry/jerry-ext/handle-scope

LaszloLango, 2019-01-31:
Hi @legendecas, thanks for bringing up this topic. We have been started to investigate the N-API, we'd like to use it in IoT.js to load [iotivity-node](https://github.com/intel/iotivity-node). I know that scope handling is needed for N-API. Your naive implementation is a good start, but probably not the most efficient way. Would you like to work on the implementation of this feature? You can do it and open a PR, if you have time to work on it and update it based on the requests of our reviewers.

legendecas, 2019-02-10:
@LaszloLango sorry for the late response. 

Yes I would definitely like to send the PR :)

PR #2753 (2019-03-27): Implement handle scope API in jerry-ext
close #2732. 

Picked from https://github.com/yodaos-project/ShadowNode/tree/master/deps/jerry/jerry-ext/handle-scope

JerryScript-DCO-1.0-Signed-off-by: legendecas legendecas@gmail.com
closed 2019-03-27 by zherczeg