jerryscript-project/jerryscript #2735: did jerryscript support "multi-threaded" or "coroutine" mechanism？

13824125580, 2019-01-28:
as i know, java supports multi-thead, and lua support coroutine mechanism, so,
did the jerry vm implemetation supports the mechanism like corotine or native supported multi-threaded? 
thanks for your support.

akosthekiss, 2019-01-28:
- JavaScript 5.1 has no thread concept, so JerryScript is single-threaded by default. 
- In JavaScript 2015 there are [promises](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Promise), which encapsulate asynchronous operations; those are supported by JerryScript if compiled with the [es2015-subset profile](https://github.com/pando-project/jerryscript/tree/master/jerry-core/profiles). 
- Moreover, JerryScript has support for external contexts, so an embedder application can use the JerryScript engine (as a library) to execute multiple JS in parallel in different contexts but that needs [port implementation](https://github.com/pando-project/jerryscript/blob/master/jerry-core/include/jerryscript-port.h#L169) to return the right context whenever it's needed (plus, currently there is no standard way of communication between the contexts).

akosthekiss, 2019-06-28:
There was no feedback from the poster, so I hope the answer was sufficient. Closing.

closed 2019-06-28 by akosthekiss