jerryscript-project/jerryscript #2745: Are there any plans to support Set() defined by ES6?

acoinfo, 2019-01-31:
Are there any plans to support Set() defined by ES6?
And report a bug: Map() did not support forEach() method.

LaszloLango, 2019-01-31:
CC: @rerobika

acoinfo, 2019-01-31:
And 'let' is a very necessary keyword, I recommend implementation ^_^

akosthekiss, 2019-01-31:
@acoinfo Issues should be focused. Right now, this issue seems to contain
- a question ("Are there any plans to support Set() defined by ES6?"),
- a bug report (?) ("Map() did not support forEach() method."), and
- a feature request ("'let' is a very necessary keyword, I recommend implementation").

It's a lot easier to conduct discussions and reach conclusions if an issue is about a single topic. So, I'd suggest you to open separate issues for unrelated questions/feedbacks.

Moreover,
- for bug reports, it is strongly preferred to include various details, like the version (git hash) the bug is reported against; the way the project is configured, built, and executed; the platform used to build and/or run the engine; a reproduction case (js file); and how the bug manifests itself; while
- for feature requests, it is good practice to provide motivation and/or use cases that justify the proposal.

(Note: Actually, the bug report here is more like a feature request. The lack of support for part of the ES6/2015 standard is not a bug, as the project explicitly states that it implements only a subset of the ES6/2015 standard at the moment.)

rerobika, 2019-02-01:
@acoinfo Despite of having a separate issue for each question would be great let me give you a quick answer.

- The Set builtin probably will be supported, but first we need to figure out how to combine the implementation with the Map builtin to reduce the code duplication.
- It is not a bug, it's a feature request. Please open an issue for it.
- `let` and `const` is more harder nut to crack. I'm working on it but it won't be published until their performance/memory usage produce similar results to what can be achieved via using `var`s.

rerobika, 2019-05-14:
@acoinfo https://github.com/pando-project/jerryscript/pull/2841 has landed, so now you can use the Set builtin object, also the missing Map.prototype.foreach routine as well.

closed 2019-05-14 by rerobika

cup, 2019-09-26:
https://github.com/jerryscript-project/jerryscript/issues/3169