jerryscript-project/jerryscript #2749: Question: How to remove dynamic function snapshot completely?

bryanle92, 2019-02-02:
I found a memory leak when loading and releasing dynamic function snapshot. My code looks like that:

```
jerry_value_t return_value;

return_value = jerry_load_function_snapshot(snapshot, snapshot_size, 0, JERRY_SNAPSHOT_EXEC_ALLOW_STATIC);
// The current allocated heap increase 3984 bytes

jerry_release_value(return_value);
jerry_gc(JERRY_GC_SEVERITY_HIGH);
// The current allocated heap decrease 2224 bytes

return_value = jerry_load_function_snapshot(snapshot, snapshot_size, 0, JERRY_SNAPSHOT_EXEC_ALLOW_STATIC);
// The current allocated heap increase 2112 bytes

jerry_release_value(return_value);
jerry_gc(JERRY_GC_SEVERITY_HIGH);
// The current allocated heap decrease 2224 bytes
```

The issue was not happened if I use static snapshot instead of dynamic snapshot (the heap always increases and decreases by exactly 24 bytes). I just want to know where does this leak come from, and is there anyway to remove a loaded dynamic snapshot completely?

LaszloLango, 2019-02-04:
@bryanle92 hi, it is probably caused by the string literals, the GC only frees the bytecode, but leaves the literals in the string table. I have to check it, but this is my thought.

bryanle92, 2019-02-04:
@LaszloLango Yes I think so. Because the leak size is somehow depended on the number of literals used in this snapshot. Do you know how to clean unused literals after the snapshot is released?

LaszloLango, 2019-02-04:
@bryanle92 cleanup of unused literals is not supported. There is no reference counting for literals at the moment, so the information about the usage is missing.

bryanle92, 2019-04-26:
@LaszloLango Hi, I just want to know do we have the plan to address this issue?

LaszloLango, 2019-04-26:
@bryanle92 hi, no we don't.

akosthekiss, 2019-06-28:
This issue is still open but discussion is inactive. It should be concluded whether it should be closed as WONTFIX/CANTFIX (if it is completely against the current architecture of the engine) or flagged as a feature request (if literal cleanup is conceptually possible only not planned in the short term).

@LaszloLango @zherczeg Opinions?

BTW, is this issue specific to snapshots? What if JS scripts are loaded and released that way?

LaszloLango, 2019-07-01:
IMHO, this is a bit difficult to implement. It is a reasonable feature request, I'd leave it open.