jerryscript-project/jerryscript #2763: Building for Particle Argon

kreneskyp, 2019-02-19:
I'm attempting to build for the Particle Argon. Jerryscript compiles but I get an error while linking.

```
/tmp/cczdc53K.ltrans0.ltrans.o: In function `jerry_fatal':
/wrk/jerryscript/jerry-core/jrt/jrt-fatals.c:58: undefined reference to `jerry_port_fatal'
```

I'm using a modified copy of  `targets/particle/Makefile.particle` to get this going:
* Updated `EXT_CFLAGS` to match  Particle's firmware [v0.8.0-rc.27 makefile](https://github.com/particle-iot/device-os/blob/v0.8.0-rc.27/build/arm-tools.mk#L30-L51).
    ```
    EXT_CFLAGS += -g3 -gdwarf-2 -Os -mcpu=cortex-m4 -mthumb -mabi=aapcs
    EXT_CFLAGS += -mfloat-abi=hard -mfpu=fpv4-sp-d16 -flto
    ```

* Updated `PLATFORM`, added `ARM_CPU`, and removed `LDFLAGS`.
   ```
    PARTICLE_BUILD_CONFIG = \
	 INCLUDE_DIRS=$(JERRYDIR)/jerry-core/include \
	 LIBS=jerry-core \
	 PLATFORM=argon \
	 LIB_DIRS=$(BUILD_DIR)/lib \
	 APPDIR=$(JERRYDIR)/targets/particle/source \
	 TARGET_FILE=jerry_main \
	 TARGET_DIR=$(BUILD_DIR) \
	 ARM_CPU=cortex-m4
    ```

* Updated jerrycore build options:

    ```
    -DCMAKE_SYSTEM_PROCESSOR=cortex-m4
    ```

LaszloLango, 2019-02-19:
@kreneskyp, there are some platform related system calls which might be different. so we introduced the port API (see: http://jerryscript.net/port-api/). There is a default implementation of it which should work on most Unix like systems. I'm not sure how do you try to build, but `JERRY_PORT_DEFAULT` is not set by default. Could you check the output of your cmake command? Or do you skip the projects build system and make your own?

LaszloLango, 2019-02-19:
It looks like to me that you are trying to include the source files only into a custom build system. I cannot recommend that. The configuration of the project will be wrong, because the cmake command does not set up the appropriate defines you need.

kreneskyp, 2019-02-19:
> there are some platform related system calls which might be different. so we introduced the port API (see: http://jerryscript.net/port-api/). There is a default implementation of it which should [...]

I didn't implement port api or set `JERRY_PORT_DEFAULT`, so that's probably it. 

Is there a reference implementation for ARM? The Particle boards I have are Cortex-M4 with ARMv7-M Thumb instruction set.

> Or do you skip the projects build system and make your own?

Here is my full Makefile.  It is a modified version of [targets/particle/Makefile.particle](https://github.com/pando-project/jerryscript/blob/master/targets/particle/Makefile.particle) 

It builds jerryscript first using jerryscript's build system. It then calls the Particle build system with settings for jerryscript's headers and static library. I've tested the second step successfully with a another static library.

```
PARTICLE_FIRMWARE := $(CURDIR)/../particle/firmware

JERRYDIR ?= $(CURDIR)
JERRYHEAP  ?= 16

BUILD_DIR  ?= $(JERRYDIR)/build/particle

# From device-os/build/arms-tools.mk
EXT_CFLAGS += -g3 -gdwarf-2 -Os -mcpu=cortex-m4 -mthumb -mabi=aapcs
EXT_CFLAGS += -mfloat-abi=hard -mfpu=fpv4-sp-d16 -flto

.PHONY: jerrycore jerry-main flash clean

PARTICLE_BUILD_CONFIG = \
	 INCLUDE_DIRS=$(JERRYDIR)/jerry-core/include \
	 LIBS=jerry-core \
	 PLATFORM=argon \
	 LIB_DIRS=$(BUILD_DIR)/lib \
	 APPDIR=$(JERRYDIR)/targets/particle/source \
	 TARGET_FILE=jerry_main \
	 TARGET_DIR=$(BUILD_DIR) \
	 ARM_CPU=cortex-m4

all: jerrycore jerry-main

jerrycore:
	mkdir -p $(BUILD_DIR)
	cmake -B$(BUILD_DIR) -H./ \
	 -DCMAKE_SYSTEM_NAME=MCU \
	 -DCMAKE_SYSTEM_PROCESSOR=cortex-m4 \
	 -DCMAKE_C_COMPILER=arm-none-eabi-gcc \
	 -DCMAKE_C_COMPILER_WORKS=TRUE \
	 -DENABLE_LTO=ON \
	 -DENABLE_ALL_IN_ONE=OFF \
	 -DJERRY_LIBM=OFF \
	 -DJERRY_CMDLINE=OFF \
	 -DFEATURE_PROFILE=es5.1 \
	 -DENABLE_STRIP=OFF \
	 -DEXTERNAL_COMPILE_FLAGS="$(EXT_CFLAGS)" \
	 -DMEM_HEAP_SIZE_KB=$(JERRYHEAP)

	make -C$(BUILD_DIR) jerry-core

jerry-main: jerrycore
	$(PARTICLE_BUILD_CONFIG) make -C$(PARTICLE_FIRMWARE) -f $(PARTICLE_FIRMWARE)/makefile

flash:
	$(PARTICLE_BUILD_CONFIG) make -C$(PARTICLE_FIRMWARE)/main -f $(PARTICLE_FIRMWARE)/main/makefile program-dfu

clean:
	$(PARTICLE_BUILD_CONFIG) make -C$(PARTICLE_FIRMWARE) -f $(PARTICLE_FIRMWARE)/makefile clean
	rm -rf $(BUILD_DIR)
```

LaszloLango, 2019-02-20:
@kreneskyp I wanted to check the build, but the particle firmware does not build for me. I tried the latest, v1.0.1 and v1.0.0.

kreneskyp, 2019-02-20:
I believe v0.8.0-rc.27 is the latest version that supports argon boards.

LaszloLango, 2019-02-20:
Side note: we only had Photon boards, so the `particle` target was tested on that.

LaszloLango, 2019-02-20:
@kreneskyp could you provide a full build log (e.g: on gist.github.com)?

galpeter, 2019-02-20:
@kreneskyp You can try using our "default" implementation of the jerry-port which is used on other ARM systems. Just call the `make jerry-port-default` after the jerry-core build and there should be an extra `libjerry-port-default.a` archive which should be linked to the final binary (just like the jerry-core in your makefile).

kreneskyp, 2019-02-20:
It's working! I also had to update `LIBS="jerry-core jerry-port-default"`.  

I have a photon, argon, and xenon. I've only tested the argon so far.

akosthekiss, 2019-06-28:
Closing as it seems that the issue is solved.

@kreneskyp We highly encourage (and appreciate) contributions. So, if you have a build setup that works, you might consider opening a PR (e.g., to patch up the particle target).

closed 2019-06-28 by akosthekiss