jerryscript-project/jerryscript #2791: Performing C SOURCE FILE Test HAVE_TM_GMTOFF failed

YaoHouyou, 2019-03-06:
Jerry version:

    Checked revisio: 85e0509
OS:

    ubuntu-14.04.5-desktop-amd64

Build command: 
    
    python tools/build.py

There are two errors show in the file `/jerryscript/build/CMakeFiles/CMakeError.log` as following:

    Performing C SOURCE FILE Test HAVE_TM_GMTOFF failed with the following output:

    Change Dir: /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp

    Run Build Command:/usr/bin/make "cmTryCompileExec1239510574/fast"
    /usr/bin/make -f CMakeFiles/cmTryCompileExec1239510574.dir/build.make CMakeFiles/cmTryCompileExec1239510574.dir/build
    make[1]: Entering directory `/home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp'
    /usr/bin/cmake -E cmake_progress_report /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp/CMakeFiles 1
    Building C object CMakeFiles/cmTryCompileExec1239510574.dir/src.c.o
    /usr/bin/cc  -D_BSD_SOURCE -D_DEFAULT_SOURCE -flto -fno-fat-lto-objects -std=c99 -pedantic -fno-builtin -fno-stack-protector -Wall -Werror=all -Wextra -Werror=extra -Wformat-nonliteral -Werror=format-nonliteral -Winit-self -Werror=init-self -Wconversion -Werror=conversion -Wsign-conversion -Werror=sign-conversion -Wformat-security -Werror=format-security -Wmissing-declarations -Werror=missing-declarations -Wshadow -Werror=shadow -Wstrict-prototypes -Werror=strict-prototypes -Wundef -Werror=undef -Wold-style-definition -Werror=old-style-definition -Wno-stack-protector -Wno-attributes -Werror -Wlogical-op -Werror=logical-op -DHAVE_TM_GMTOFF -Wno-error=strict-prototypes -Wno-error=old-style-definition   -o CMakeFiles/cmTryCompileExec1239510574.dir/src.c.o   -c /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp/src.c
    /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp/src.c:4:5: warning: function declaration isn’t a prototype [-Wstrict-prototypes]
     int main()
     ^
    /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp/src.c: In function ‘main’:
    /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp/src.c:4:5: warning: old-style function definition [-Wold-style-definition]
    /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp/src.c:7:4: error: statement with no effect [-Werror=unused-value]
    tmp->tm_gmtoff;
    ^
    cc1: all warnings being treated as errors
    make[1]: *** [CMakeFiles/cmTryCompileExec1239510574.dir/src.c.o] Error 1
    make[1]: Leaving directory `/home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp'
    make: *** [cmTryCompileExec1239510574/fast] Error 2
    
    Source file was:
    
    #include <time.h>
    
    int main()
    {
       struct tm* tmp;
       tmp->tm_gmtoff;
      return 0;
    }

Maybe for this reason, i have same problem as [issue](https://github.com/pando-project/jerryscript/issues/2193). Although i tried the solution, it still failed.

LaszloLango, 2019-03-06:
This looks like it is independent from the engine, I don't see any JerryScript related code in your source.

galpeter, 2019-03-06:
Could you please share the full build.py output? The `HAVE_TM_GMTOFF` test is generated by the cmake build system. What is your cmake version?

YaoHouyou, 2019-03-06:
@galpeter 

```
nisl@nisl-H81M-DS2:~$ cmake --version
cmake version 2.8.12.2
```
There are only two log files, the other log file is `jerryscript/CMakeFiles/CMakeOutput.log`, and the content is:  
```
The system is: Linux - 4.4.0-142-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags: 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/nisl/yhy/jerryscript/build/CMakeFiles/2.8.12.2/CompilerIdC/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec3662415559/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec3662415559.dir/build.make CMakeFiles/cmTryCompileExec3662415559.dir/build
make[1]: Entering directory `/home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec3662415559.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec3662415559.dir/testCCompiler.c.o   -c /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTryCompileExec3662415559
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec3662415559.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec3662415559.dir/testCCompiler.c.o  -o cmTryCompileExec3662415559 -rdynamic 
make[1]: Leaving directory `/home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec3363091924/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec3363091924.dir/build.make CMakeFiles/cmTryCompileExec3363091924.dir/build
make[1]: Entering directory `/home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec3363091924.dir/CMakeCCompilerABI.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec3363091924.dir/CMakeCCompilerABI.c.o   -c /usr/share/cmake-2.8/Modules/CMakeCCompilerABI.c
Linking C executable cmTryCompileExec3363091924
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec3363091924.dir/link.txt --verbose=1
/usr/bin/cc     -v CMakeFiles/cmTryCompileExec3363091924.dir/CMakeCCompilerABI.c.o  -o cmTryCompileExec3363091924 -rdynamic  
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.4' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.4) 
COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'cmTryCompileExec3363091924' '-rdynamic' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/4.8/collect2 --sysroot=/ --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu --as-needed -export-dynamic -dynamic-linker /lib64/ld-linux-x86-64.so.2 -z relro -o cmTryCompileExec3363091924 /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/crt1.o /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/4.8/crtbegin.o -L/usr/lib/gcc/x86_64-linux-gnu/4.8 -L/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/4.8/../../.. CMakeFiles/cmTryCompileExec3363091924.dir/CMakeCCompilerABI.c.o -lgcc --as-needed -lgcc_s --no-as-needed -lc -lgcc --as-needed -lgcc_s --no-as- [... cut ...] or=missing-declarations -Wshadow -Werror=shadow -Wstrict-prototypes -Werror=strict-prototypes -Wundef -Werror=undef -Wold-style-definition -Werror=old-style-definition -Wno-stack-protector -Wno-attributes -Werror -Wlogical-op -Werror=logical-op -Wno-error=strict-prototypes -Wno-error=old-style-definition    CMakeFiles/cmTryCompileExec1346210715.dir/CheckIncludeFiles.c.o  -o cmTryCompileExec1346210715 -rdynamic 
make[1]: Leaving directory `/home/nisl/yhy/jerryscript/build/CMakeFiles/CMakeTmp'
```

galpeter, 2019-03-06:
What I'm interested in is the output of the `python tools/build.py` call.
I've just checked on an ubuntu 14.04.5 and the build was ok, no errors for me.
The log you copied here is ok.

akosthekiss, 2019-06-28:
@YaoHouyou Could you please provide the log that was requested by @galpeter ? Or has the issue already been solved? (Feedback is appreciated.)

YaoHouyou, 2019-06-29:
@galpeter @akosthekiss I’m sorry for that the `Ubuntu 14.04` has been deleted. This problem has been resolved in higher versions of the operating system, such as `Ubuntu 16.04`, `Ubuntu 18.04`. Thank you again for your help. If i meet this problem again, i am very willing to provide detailed information about the log.

ossy-szeged, 2020-10-16:
Still valid issue and can be easily fixed.

PR #4292 (2020-10-17): Fix HAVE_TM_GMTOFF check for older CMake.
Older CMake (for example 2.8.12.2) generates feature check with unused
variable, which reports false positive fail due to -Werror=unused-value.
The feature is present and works, so we should only ignore this build warning.

Fixes #2791.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác csaba.osztrogonac@h-lab.eu
commit 9da1211e (2020-10-17): Fix HAVE_TM_GMTOFF check for older CMake. (#4292)

Older CMake (for example 2.8.12.2) generates feature check with unused
variable, which reports false positive fail due to -Werror=unused-value.
The feature is present and works, so we should only ignore this build warning.

Fixes #2791.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác csaba.osztrogonac@h-lab.eu
closed 2020-10-17 by ossy-szeged