jerryscript-project/jerryscript #2799: Nominating Robert Fancsik (@rerobika) to become a JerryScript reviewer

zherczeg, 2019-03-19:
I would like to nominate Robert Fancsik to become a JerryScript reviewer. I trust him to follow our process. He has already 60 informal reviews:

#2793 Rework usages/naming of configuration macros
#2796 Increase branch coverage: Array.prototype functions
#2797 Split the Travis checks into two test calls
#2789 Make jerryx_port_handler_print_char truly a port function
#2715 [Proposal] Introduce new jerry-ext methods to ease method/property registration
#2753 Implement handle scope API in jerry-ext
#2764 Separate the debugger-client to support the later usage of other communication protocols
#2599 Implement ES2015 module system
#2754 Compile with clang-7.0
#2759 Fix JSON.stringify
#2752 Allow build artifacts on AppVeyor
#2751 Added 'instanceof' binary operation to the API
#2633 Support Bluetooth communcation JerryScript debugger
#2723 Added new 'jerry_value_instanceof' API function.
#2746 Added new 'jerry_binary_operation' API function
#2717 Increase test coverage: Array.prototype.toString
#2725 Added new 'jerry_value_strict_equal' API function.
#2682 Increase test coverage: Array.prototype.splice
#2722 Fixed unreachable jump statements.
#2716 Increase test coverage: Array.prototype.toLocaleString
#2718 Use union to convert between jerryx_arg_int_option_t and uintptr_t
#2674 Increase test coverage: Array.prototype.sort
#2680 Add pkg-config metadata files for libraries
#2714 Fix out-of-bounds writes (buffer overflows) in unit tests
#2710 Fix buffer overflow while printing unhandled exception
#2707 Use logical operators for bool converions.
#2706 Fix some type conversions in ecma-builtins.
#2684 Update the debugger to work on Windows
#2705 Style fix in the lexer.h
#2703 Fix api example in the documentation
#2697 Bump ESP8266 target to SDK v2.1.0
#2685 Increase test coverage: Array.prototype.indexOf
#2688 Increase test coverage: Array.prototype.lastIndexOf
#2543 Add RegExp recursion depth limit
#2634 Fix named function expression creation.
#2628 Fix floating number parsing.
#2578 Add new test
#2571 Add support for function argument initializers.
#2567 Support methods for object initializers.
#2572 Fix type conversion errors in case of TizenRT.
#2540 Change jerry_port interface to allow for a correct implementation of timezones.
#2550 Remove the JERRY_SNAPSHOT_FOUR_BYTE_CPOINTER snapshot flag.
#2547 Rework op_construct.
#2481 Implement computed properties for object literals.
#2432 Support for shorthand object literal notation.
#2447 Implement the core of the map object.
#2442 Fix a few parser style issues.
#2440 Fix for null pointer dereference in jmem_heap_free_block
#2419 Migration Guide
#2406 Splitting the debugger and console part of the python debugger
#2430 Reduce "with" stack consumption by 1.
#2424 Fixes for ES2015 classes.
#2414 Rework function call.
#2410 Add %TypedArray%.prototype.subarray([ begin [, end ] ]) support.
#2350 Merge jerry_get_value_without_error and jerry_value_clear_error_flag
#2275 Reduce code duplication between Array.forEach, some, and every
#2255 Implement toString and join for TypedArrays.
#2214 Modified ecma string to utf8 string conversion to reduce binary size.
#2165 Add TypedArray C API
#2144 Bugfix: add missing stdlib.h include for rand() usage in ecma-builtin-math.c

Seconds are welcome.

DanielBallaSZTE, 2019-03-19:
I'm not a reviewer/committer, however I do second him getting nominated.

akosthekiss, 2019-03-27:
Five seconds and no objections. Congrats to our new reviewer, @rerobika !

closed 2019-03-27 by akosthekiss