jerryscript-project/jerryscript #2802: ecma_utf8_string_to_number error for "2e23"

libin049, 2019-03-22:
Function `ecma_utf8_string_to_number（"2e23",4） `  result is  200000000000000016777216
''. It is not correct.  The correct result is 199999999999999983222784

LaszloLango, 2019-03-22:
@libin049 `ecma_utf8_string_to_number` is an internal function, could you provide an example which triggers this issue via API calls? Or a JS example?

akosthekiss, 2019-03-22:
Could you please explain why do you think that "2e+23 +16777216" is a more correct result for 2e+23 than "2e+23 -16777216"?

Note: I'm inclined to believe that this is a duplicate of #2638 and as such, invalid. `e+23` is just too big to be precisely represented in an IEEE 754 `double`.

libin049, 2019-03-23:
@LaszloLango, thers is not JS example. I want to implement "strtod" using this function.
@akosthekiss, I am not sure "2e+23 +16777216" is more correct result than "2e+23 -16777216".  However, I found both gcc(v5.4.0) and clang(v5.0.2) get "2e+23 -16777216".

dbatyai, 2019-03-23:
@akosthekiss This is in fact a valid issue, as the current result does not comply with the rounding rules set by the standard.

The default rounding rule for binary floating point numbers should be **"Round to nearest, ties to even"**, which means that for numbers that fall exactly halfway between two representations, the one that has an even least significant digit (in binary) should be chosen.

In the case of the "2e+23 +16777216" representation the least significant byte is `11110111`, so as by the standard "2e+23" should be rounded down.

PR #2890 (2019-05-31): Fix rounding issue in ecma_utf8_string_to_number
Fixes #2802

JerryScript-DCO-1.0-Signed-off-by: Daniella Barsony bella@inf.u-szeged.hu
commit b6fc4e13 (2019-05-31): Fix rounding issue (#2890)

Fixes #2802

JerryScript-DCO-1.0-Signed-off-by: Daniella Barsony bella@inf.u-szeged.hu
closed 2019-05-31 by LaszloLango