jerryscript-project/jerryscript #2812: Jerryscript not build for NRF52840 target

ARM93, 2019-04-04:
I have tried to build the Jerryscript mbedOS for NRF52840 platform by using the command "gulp --target=NRF52840_DK". But i have have faced the "core_cm4.h" header file not visible to "nrf52840.h" file.

So i have copied the cmsis path header files into "./build/jerryscript/targets/mbedos5/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/device/" path and then the "core_cm4.h" file issue get resolved. 

Now i have faced below error,
/repel/mbed-js-repl-example$ gulp --target=NRF52840_DK
[17:14:11] Using gulpfile ~/repel/mbed-js-repl-example/Gulpfile.js
[17:14:11] Starting 'make-build-dir'...
[17:14:11] Finished 'make-build-dir' after 135 μs
[17:14:11] Starting 'makefile'...
[17:14:11] Starting 'bundle'...
[17:14:11] Starting 'ignorefile'...
[17:14:11] Finished 'ignorefile' after 25 ms
[17:14:11] Finished 'makefile' after 44 ms
[17:14:11] Starting 'get-jerryscript'...
[17:14:11] Finished 'get-jerryscript' after 116 μs
[17:14:11] Starting 'getlibs'...
[17:14:11] Finished 'getlibs' after 12 ms
$ make getlibs
make: Nothing to be done for 'getlibs'.
[17:14:11] Finished 'bundle' after 99 ms
[17:14:11] Starting 'cppify'...
[17:14:11] Finished 'cppify' after 25 ms
[17:14:11] Starting 'build'...
[17:14:12] Found native packages: mbed-js-repl
$ make BOARD=NRF52840_DK EXTRAS=/home/pc/repel/mbed-js-repl-example/node_modules/mbed-js-repl/.
[17:14:13] build/make
[17:14:13] Finished 'build' after 1.72 s
[17:14:13] Starting 'default'...
[17:14:13] Finished 'default' after 29 μs
cd jerryscript/targets/mbedos5 && pip install -r tools/requirements.txt && rm -f source/js_encoded.cpp && rm -f source/pins.cpp && rm -f source/main.cpp && make BOARD=NRF52840_DK EXTRA_SRC="/home/pc/repel/mbed-js-repl-example/node_modules/mbed-js-repl/.:../../../source" EXTERN_BUILD_DIR=../../../out/NRF52840_DK NO_JS=1
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: pycparser<=2.17 in /usr/local/lib/python2.7/dist-packages (from -r tools/requirements.txt (line 1)) (2.17)
Requirement already satisfied: simpleeval in /usr/local/lib/python2.7/dist-packages (from -r tools/requirements.txt (line 2)) (0.9.8)
Requirement already satisfied: pycparserext in /usr/local/lib/python2.7/dist-packages (from -r tools/requirements.txt (line 3)) (2016.2)
Requirement already satisfied: ply>=3.4 in /usr/local/lib/python2.7/dist-packages (from pycparserext->-r tools/requirements.txt (line 3)) (3.11)
make[1]: Entering directory '/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5'
python tools/generate_pins.py NRF52840_DK
Traceback (most recent call last):
  File "tools/generate_pins.py", line 247, in <module>
    main()
  File "tools/generate_pins.py", line 237, in main
    pins = enumerate_pins(pins_file, ['./tools'] + list(includes), defines)
  File "tools/generate_pins.py", line 154, in enumerate_pins
    parser=GnuCParser())
  File "/usr/local/lib/python2.7/dist-packages/pycparser/__init__.py", line 93, in parse_file
    return parser.parse(text, filename)
  File "/usr/local/lib/python2.7/dist-packages/pycparserext/ext_c_parser.py", line 64, in parse
    return self.cparser.parse(text, lexer=self.clex, debug=debuglevel)
  File "/usr/local/lib/python2.7/dist-packages/pycparser/ply/yacc.py", line 331, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/usr/local/lib/python2.7/dist-packages/pycparser/ply/yacc.py", line 1106, in parseopt_notrack
    p.callable(pslice)
  File "/usr/local/lib/python2.7/dist-packages/pycparser/c_parser.py", line 654, in p_decl_body
    typedef_namespace=True)
  File "/usr/local/lib/python2.7/dist-packages/pycparser/c_parser.py", line 394, in _build_declarations
    self._parse_error('Invalid declaration', coord)
  File "/usr/local/lib/python2.7/dist-packages/pycparser/plyparser.py", line 55, in _parse_error
    raise ParseError("%s: %s" % (coord, msg))
pycparser.plyparser.ParseError: /home/pc/gcc-arm-none-eabi-6_2-2016q4/arm-none-eabi/include/sys/_stdint.h:48: Invalid declaration
Makefile:72: recipe for target 'source/pins.cpp' failed
make[1]: Leaving directory '/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5'
make[1]: *** [source/pins.cpp] Error 1
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: Command `make BOARD=NRF52840_DK EXTRAS=/home/pc/repel/mbed-js-repl-example/node_modules/mbed-js-repl/.` exited with code 2
    at ChildProcess.handleSubShellExit (/home/pc/repel/mbed-js-repl-example/node_modules/gulp-run/command.js:166:13)
    at Object.onceWrapper (events.js:317:30)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)

i have attached "_stdint.h" for reference[_stdint.h.zip](https://github.com/pando-project/jerryscript/files/3043344/_stdint.h.zip)
From the file i understood the __uint32_t is invalid. Could you help to resolve this issue. 

Please note the same source successfully compiled for NRF52832 by using the "NRF52_DK" command.

Environment details:
Host OS: Ubuntu 16.04 LTS
Target : nRF52840
CC Version: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1 20161205
Jerryscript Source Repository: https://github.com/ARMmbed/mbed-js-repl-example

ARM93, 2019-04-06:
Can anybody explain why this error come in **_stdint.h** file even it was defined in "**#include <machine/_default_types.h>**" header file?

ARM93, 2019-04-09:
Can anybody help to figure out this issue?

LaszloLango, 2019-04-09:
CC: @fbmrk @thegecko

ARM93, 2019-04-12:
I have done the jerryscript porting for NRF52840 chipset by modified the PinNames.h manually **(i have Only configured port 0 pins in PinNames.h file. if add the port 1 pins in PinNames.h file then faced some compile time issues. So i have skipped the port 1 pins configuration in PinNames.h file)** and did some patch work in definitions.json file.

After long struggle then the hex/bin file generated & it booted successfully in NRF52840 platform. By using the main.js file i can able control the LEDs. But i faced the issues in JerryScript console.

Once after the JerryScript console get opened then press any keys in keyboard (**except the "Enter" key**) then i faced below error.

[image]

thegecko, 2019-04-12:
Cc @janjongboom

janjongboom, 2019-04-15:
@ARM93 The ISR issue is fixed in mbed-js-v2 like this: https://github.com/janjongboom/mbed-js-repl/commit/28b996d9c902223fbae4d0e67f2749c8d8117293 Should be able to land this cleanly on mbed-js-repl.

Unfortunately pin name generation still breaks on the NRF52840_DK target.

ARM93, 2019-04-15:
> @ARM93 The ISR issue is fixed in mbed-js-v2 like this: [janjongboom/mbed-js-repl@28b996d](https://github.com/janjongboom/mbed-js-repl/commit/28b996d9c902223fbae4d0e67f2749c8d8117293) Should be able to land this cleanly on mbed-js-repl.
> 
> Unfortunately pin name generation still breaks on the NRF52840_DK target.

@janjongboom Thanks for your update. Now console is working fine. :-)

Now i start to working on port 1 adding into PinNames.h header file issue.

janjongboom, 2019-04-19:
@ARM93 Pin generation is fixed now in https://github.com/ARMmbed/mbed-js-v2/commit/a41fce512ea8eb9692ef3374548db1ef4ffa8bb4. I'm inclined not to upstream this to this repository as we won't maintain the target here in the future, but rather support mbed-js-v2.

ARM93, 2019-04-19:
@janjongboom Thanks for your support.  This  **ARMmbed/mbed-js-v2@a41fce5** update not resolved my issue. I just added the port 1 pins as port 0 pin in the PinName enum then my issue getting resolved.
Find the attached file for your reference [PinNames.h.zip](https://github.com/pando-project/jerryscript/files/3098456/PinNames.h.zip)

closed 2019-04-19 by ARM93