jerryscript-project/jerryscript #2815: async/await support

PaulFreund, 2019-04-09:
Hi there,

as jerryscript already has support for Promises I was curious how much work it would be to support async/await. Currently I am looking into using generators but as generators seem to create relatively simple code I was wondering if this was achievable with a moderate amount of work.

Best regards,

Paul

LaszloLango, 2019-04-09:
@PaulFreund Unfortunately we do not have any plan to implement async/await at the moment. However new contributions are always welcome. :) Feel free to open a PR if you have a working solution.

PaulFreund, 2019-04-09:
@LaszloLango That is good to hear. I will evaluate the possibility but first would be very happy to get a small estimation from you or some other maintainer if the technology is compatible with the current architecture of jerryscript. I'm not a heavy user yet but so far I see it as the best implementation for embedded out there.

zherczeg, 2019-04-09:
I don't see a major blocker to do it in JerryScript, but it is not a trivial task though. Function stack should be kept in memory to allow suspending / resuming execution.

zherczeg, 2020-05-27:
The parsing part is completed for this construct, and the operation part is in progress.

PaulFreund, 2020-06-22:
Wow! Thank you very much for all involved, is this reopened because of #3916?

dbatyai, 2020-06-22:
Basically yes, I thought we should keep this open until the remaining asnyc features are implemented.

closed 2020-08-27 by dbatyai