jerryscript-project/jerryscript #2817: Find a Segmentation fault bug

yeguixin, 2019-04-11:
### Jerry Version:

   Checked revision: 40f7b1c
   Build: debug.linux

### OS:

   Ubuntu 16.04,   x86_64

### Testcase:

`var a = function(f) {
    function f() {}
    function f() {}
    return arguments;
};
a(6);`

### Backtrace:

execve("build/bin/jerry", ["build/bin/jerry", "testcase1.js"], [/* 73 vars */]) = 0
brk(NULL)                               = 0x1818000
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=100849, ...}) = 0
mmap(NULL, 100849, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fca08a9a000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P\t\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1868984, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fca08a99000
mmap(NULL, 3971488, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fca084c4000
mprotect(0x7fca08684000, 2097152, PROT_NONE) = 0
mmap(0x7fca08884000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1c0000) = 0x7fca08884000
mmap(0x7fca0888a000, 14752, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fca0888a000
close(3)                                = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fca08a98000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fca08a97000
arch_prctl(ARCH_SET_FS, 0x7fca08a98700) = 0
mprotect(0x7fca08884000, 16384, PROT_READ) = 0
mprotect(0x633000, 4096, PROT_READ)     = 0
mprotect(0x7fca08ab3000, 4096, PROT_READ) = 0
munmap(0x7fca08a9a000, 100849)          = 0
brk(NULL)                               = 0x1818000
brk(0x1839000)                          = 0x1839000
open("testcase1.js", O_RDONLY)          = 3
fstat(3, {st_mode=S_IFREG|0664, st_size=147, ...}) = 0
read(3, "var a = function(f) {\n    printf"..., 1048576) = 147
read(3, "", 1044480)                    = 0
close(3)                                = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x87a5820} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

rerobika, 2019-04-11:
@yeguixin Could you please check the related test case again after applying #2816 ? I think it should solve this problem.

galpeter, 2019-04-11:
Based on the test case this is the same as #2699 and I've just tested that the PR #2816 fixes this segfault.

Also please note that the backtrace mentioned only shows the system/library calls (I guess it was created via strace/ltrace), thus it is not a real backtrace.

akosthekiss, 2019-04-11:
duplicate of #2699

closed 2019-04-11 by akosthekiss