jerryscript-project/jerryscript #2826: The size of snapshot from "jerry_generate_snapshot".

Becavalier, 2019-04-15:
what does the "size of snapshot" mean here? Do you mean it's the byte length of the "buffer_p" array? Actually it must the integer times of uint32, corresponding to 4 bytes. But what I got is not.

[image]

LaszloLango, 2019-04-15:
@Becavalier Hi, the return size is the byte length of the snapshot. The reason  of the `uint32_t` output buffer is to ensure the buffer is aligned to 4 bytes. Unfortunately the documentation does not make it clear. I will update the description.

Becavalier, 2019-04-15:
@LaszloLango Thanks, I get it. So the return value is the actual bytes length the snapshot used.

closed 2019-05-24 by Becavalier