jerryscript-project/jerryscript #2832: Testcase that Jerry cannot execute

yeguixin, 2019-04-16:
### Jerry Version:

   Checked revision: 40f7b1c27fa79568deb5a473b60397ea52b787f7
   Build: debug.linux

### OS:

   Ubuntu 16.04,   x86_64

### Testcase:

[https://github.com/yeguixin/testcases/blob/master/0/0bee07592a100236e9b99c722c3388f1-0.js](url)

### Exception:
When I execute the testcase using jerry, I find that jerry cannot execute the js file. Maybe it's a bug of Jerry.

akosthekiss, 2019-04-16:
@yeguixin Could you please give some details on what "jerry cannot execute the js file" means? How did you build the project? (Please, give the exact command line.) What diagnostics are reported? What is the exit code of the program?

The first thing that occurs to me is that the test case is almost 9000 lines long and its size exceeds 500k, which may result in OOM. But more details are needed to say anything for sure.

yeguixin, 2019-04-16:
I build Jerry using python2.7 as follows:
python tools/build.py

I add print function in that testcase as you see in first line of the testcase. But jerry does not print the content in the console, which I think jerry cannot execute the testcase. I am sure that I successfully build jerry as it is work for other js files.

Maybe it's caused by OOM, I delete some code and it works.

akosthekiss, 2019-04-16:
I suggest building the project with
```bash
tools/build.py --debug --logging=on --error-messages=on
```
That may give more output.

yeguixin, 2019-04-17:
I rebuild Jerry as you suggested. When I execute that testcase, Jerry reports an error as follows:

Error: ERR_OUT_OF_MEMORY

Now the exception is clear and it's caused by OOM.

akosthekiss, 2019-04-17:
Thanks for confirming. OOM is not considered a bug.

closed 2019-04-17 by akosthekiss

LaszloLango, 2019-04-17:
*Note*: If you need more memory than 512K then you can enable 32 bit compressed pointers, like this
`python tools/build.py --cpointer-32bit=on --mem-heap=1024`

See also: http://jerryscript.net/getting-started/

yeguixin, 2019-04-17:
Thanks @LaszloLango @akosthekiss