jerryscript-project/jerryscript #2858: List of things that JerryScript can do and can't do?

IngwiePhoenix, 2019-05-03:
Hello.

By skimming the source files and the wiki as well as the project's page, I kept reading that it supports ECMAScript 5.1, however JerryScript is implemented in such a small source tree, that I wonder what it actually supports - and what not?

A small listing of what it does and does not do, would be very nice :)

Kind regards,
Ingwie

akosthekiss, 2019-05-03:
Hi @IngwiePhoenix,

My fellow maintainers will correct me if I'm wrong, but according to my best understanding ES5.1 support is indeed complete and up to the specs. ES5.1 version (branch) of test-262 is passing fine.

And yes, the source code is small (even if that's a subjective assessment), because that was the design goal of JerryScript right from the start: both code and memory footprint should be as small as possible. (Note that this comes at a price: don't expect performance boost by just-in-time compilation or anything like that available in JS engines that target "big" boxes.)

What's not complete is ES6 support (and above). And there you have a point: we still haven't published a checklist on the feature set, although requested by many for long.

I hope this answers your question,
Akos

IngwiePhoenix, 2019-05-04:
Hello @akosthekiss !

Yes, that reply helped me a lot, thank you very much :) I took Jerry to a test and just threw some random stuff at the repl to see if they'd work. Sort of like manual feature detection, I guess... :)

A list of supported features would really help a lot, because many people may know what ECMAScript means and is, but not exactly which features fall into 5.1 - let alone what ES6 features may be supported by this time.

Thank you for your answer! =)

closed 2019-05-04 by IngwiePhoenix