jerryscript-project/jerryscript #2866: Function.toString() don't not work.

cyongbin, 2019-05-10:
Function.toString() can'not return function body now. When will this feature be supported?

LaszloLango, 2019-05-13:
@cyongbin [`Function.prototype.toString()`](https://www.ecma-international.org/ecma-262/5.1/#sec-15.3.4.2) is implementation-dependent. JerryScript does not save the necessary information to restore the original function. We do not plan to support this feature because it needs too much memory and we would like to keep the memory consumption of the engine as low as possible.

cyongbin, 2019-05-13:
It is recommended to set this feature as a configurable item and turn it off when not needed.

LaszloLango, 2019-05-13:
Feel free to contribute to the project.

cyongbin, 2019-05-13:
All right. I have rewrote a web server on JerryScript, but it's not perfect yet.

acoinfo, 2019-06-28:
`Function.prototype.toString()` is still important. Many third-party JS libraries use this method. If you consider reducing memory consumption, you can make a compile configurable option.

akosthekiss, 2019-06-28:
@acoinfo As @LaszloLango mentioned, it is not on our short term roadmap to implement this feature ourselves. But we are happy to review a pull request that implements the feature (in a compile-time configurable way), should any community member contribute it.

I'll flag this as a feature request for now.

sdwvit, 2021-01-26:
hi! any suggestion on where to start digging into this feature implementation? any idea would be much appreciated