jerryscript-project/jerryscript #2867: BUG: Reference error.

cyongbin, 2019-05-10:
When I upgrade to the latest version(2019.5.9), the reference is always error.

LaszloLango, 2019-05-10:
@cyongbin hi, this report does not make any sense. Please provide more information. How did you build JerryScrript? What is your code that you would like to run? What is your system environment? Reference error is usually caused by undeclared variable or property, so why do you think this is an error in the engine and not in your JS code?

cyongbin, 2019-05-10:
@LaszloLango The last version of jerryscript I used was 2019.4.3, and I got an error when I updated to the latest version of 2019.5.9.
In the latest version, long-running iotjs complex programs (like mqtt) will crash. The trace discovery is that the reference count error caused the object to be erroneously released.

akosthekiss, 2019-05-10:
@cyongbin Please *do* provide the information requested by @LaszloLango. In its current form your issue report does not allow the bug to be reproduced. Please submit the relevant information:
- your build and execution environment: architecture and OS,
- your build command,
- a (preferably minimal) test case: JS code that triggers the problem,
- the error log (stderr and backtrace).

LaszloLango, 2019-05-10:
@cyongbin if this is an issue in IoT.js then please report the issue there with the requested information to allow us to reproduce and investigate the issue. I've just updated the JerryScript dependency in IoT.js and it works well.

acoinfo, 2019-05-10:
I also encountered a reference problem. In order to test that I blocked these recycling code, my memory is large enough.
[image]
Then loop through a js program, jerry will report the ERR_REF_COUNT_LIMIT error after a while, I printed the relevant call stack as follows:
[image]
Previous versions such as 2019-04-03 have the same application and js native library without this problem.

acoinfo, 2019-05-10:
I used malloc/free by define: JERRY_SYSTEM_ALLOCATOR
(I guess: The code mentioned above is blocked and it seems that there should be no and should not exist when JERRY_SYSTEM_ALLOCATOR is defined.)

LaszloLango, 2019-05-10:
Please provide the the requested information and the minimal example to be able to reproduce.

akosthekiss, 2019-05-10:
@acoinfo Your code seems to be modified from official version. Please provide an example that works on unmodified code base, and specify the exact git hash where the error can be observed. We *cannot* help without these details.

acoinfo, 2019-05-10:
I constructed a test case
[image]

[image]

[image]

[image]

[image]

acoinfo, 2019-05-10:
The time of the delay is 10ms, and an error occurs after about 5 minutes of operation (same as before, the allocation gc is masked)

acoinfo, 2019-05-10:
This test case is simpler!!!
[image]

‘print’ is jerry-ext jerryx_handler_print()

LaszloLango, 2019-05-10:
Please do not post screen shots, we cannot copy the code from them. Use standard markdown formatting to post code blocks in github comments or use https://gist.github.com to share code/output.

LaszloLango, 2019-05-10:
First of all this issue is raised in the wrong repository. Probably the issue comes from the wrong usage of the engine and not from the engine itself.

Secondly, if `JS_GET_ARG` is the same macro as in [IoT.js](https://github.com/pando-project/iotjs), then your are using it wrong. You must free `time_ms` with `jerry_release_value`. This won't fix your ref_count problem, but it resolves a memory leak in your code.

Thirdly, is this some modified version of [IoT.js](https://github.com/pando-project/iotjs)? If so, then please close this and open a new issue in that project.

LaszloLango, 2019-05-10:
Hint: this issue might be caused by wrong usage of `jerry_acqiure_value` and `jerry_release_value`. Check your native code to make sure you release your acquired values.

LaszloLango, 2019-05-10:
The date of the latest commit does not identify the exact version that you are using. Please provide a **commit hash** next time.

**Hint 2**:  You can try to track down the "problematic" commit with [`git bisect`](https://git-scm.com/docs/git-bisect). That can lead us closer to the root of your problem.

acoinfo, 2019-05-10:
I didn't use IoT.js. I just suggested that you can test it in the IoT.js environment. Here's a code you can try running in the jerryscript shell environment:

Use the current git master header version.  
Build jerryscript with `-DJERRY_SYSTEM_ALLOCATOR` and `-DJERRY_CPOINTER_32_BIT`  
You need to modify `jmem-allocator.c` to compile. This is another bug, not discussed here.  
Since I used `-DJERRY_SYSTEM_ALLOCATOR`, the `jmem_heap_allocated_size` and `jmem_heap_limit` in the context have lost their effect (always 0) so I need to mask the following code:  
```
static void *
jmem_heap_gc_and_alloc_block (...)
{
    ...
#if 0
  if (JERRY_CONTEXT (jmem_heap_allocated_size) + size >= JERRY_CONTEXT (jmem_heap_limit))
  {
    jmem_run_free_unused_memory_callbacks (JMEM_FREE_UNUSED_MEMORY_SEVERITY_LOW);
  }
#endif
  ...
}
```
then make jerryscript and run following code:
```
function log() {
	var out = '';
	var first = true;
	for (var i = 0; i < arguments.length; i++) {
		if (first) {
			first = false;
		} else {
			out += ' ';
		}
		out += arguments[i];
	}
	print(out);
}

while (true) {
	log('Hello World!');
}
```
fatal!!!

LaszloLango, 2019-05-13:
@acoinfo I cannot reproduce the issue.

**build command**
`tools/build.py --clean --cpointer-32bit=on --system-allocator=on --compile-flag=-m32 --external-context=on`

**build environment**:
 - Ubuntu 18.04 LTS
 - gcc 7.4

acoinfo, 2019-05-13:
OK! But my target machine is a pure 64-bit environment.
The application virtual address space I can confirm within 4GB

LaszloLango, 2019-05-13:
@acoinfo system allocator is NOT supported on 64bit systems. Do not modify the `jmem-allocator.c` and it will work as expected.

acoinfo, 2019-05-13:
But however, I use `JERRY_SYSTEM_ALLOCATOR` to use `stdlib.h` -> `malloc() / free()` and the application memory space high 32 bits is zero, which memory pointer can be converted from uint32_t, Still can't use 64 bit? (I used `JERRY_ENABLE_EXTERNAL_CONTEXT` & `JERRY_CPOINTER_32_BIT` & `JERRY_SYSTEM_ALLOCATOR`)
If I trigger gc often, there is no reference counting problem, but the following error occurs:
```
...
[62] 0x4913f8f4 (kernel@0x0+0x4913f8f4 abort+12)
[61] 0x80862408 (/lib/libvpmpdm.so@0x80850000+0x12408 dlfree+1112)
[60] 0x80106e1c (/lib/jsre/libjerry-core.so@0x800b0000+0x56e1c jmem_heap_free_block+84)
[59] 0x800cdcb4 (/lib/jsre/libjerry-core.so@0x800b0000+0x1dcb4 ecma_dealloc_extended_object+28)
[58] 0x800d93d4 (/lib/jsre/libjerry-core.so@0x800b0000+0x293d4 ecma_deref_object+3084)
[57] 0x800d974c (/lib/jsre/libjerry-core.so@0x800b0000+0x2974c ecma_gc_run+472)
[56] 0x800d98dc (/lib/jsre/libjerry-core.so@0x800b0000+0x298dc ecma_free_unused_memory+140)
[55] 0x80107160 (/lib/jsre/libjerry-core.so@0x800b0000+0x57160 jmem_run_free_unused_memory_callbacks+44)
[54] 0x80106cd4 (/lib/jsre/libjerry-core.so@0x800b0000+0x56cd4 jmem_heap_finalize+192)
[53] 0x80106da0 (/lib/jsre/libjerry-core.so@0x800b0000+0x56da0 jmem_heap_alloc_block+24)
[52] 0x80106edc (/lib/jsre/libjerry-core.so@0x800b0000+0x56edc jmem_pools_alloc+156)
[51] 0x800cdc50 (/lib/jsre/libjerry-core.so@0x800b0000+0x1dc50 ecma_alloc_object+16)
[50] 0x800d3664 (/lib/jsre/libjerry-core.so@0x800b0000+0x23664 ecma_create_decl_lex_env+16)
[49] 0x80105e44 (/lib/jsre/libjerry-core.so@0x800b0000+0x55e44 ecma_op_function_call+528)
[48] 0x801210bc (/lib/jsre/libjerry-core.so@0x800b0000+0x710bc vm_run_eval+1312)
...
```

LaszloLango, 2019-05-13:
If your allocator can guarantee that the given address is always use the lower 32bits then it could work. Why don't you simply compile as a 32bit application (with `-m32` compiler option) if you don't need the whole address space?

acoinfo, 2019-05-13:
I don't use a Linux system. I use SylixOS (A powerful RTOS that supports multiple processes similar to VxWorks) operating system running on the ARMv8 Cortex-A53 4 Cores SMP processor. The operating system and runtime libraries are all 64bit environments. I am designing an edge computing operating system that has done a lot of work based on jerryscript.

When I encountered this problem, I used the same test program on the x86 32bit system, and a similar problem occurred.

acoinfo, 2019-05-13:
Sorry, I think I have described two problems. This discussion is a reference counting problem. I am running on a 32 bit system. The compilation options are: **`JERRY_ENABLE_EXTERNAL_CONTEXT`** & **`JERRY_CPOINTER_32_BIT`** & **`JERRY_SYSTEM_ALLOCATOR`**. For example, the test code is as follows:
```
function log() {
	var out = '';
	var first = true;
	for (var i = 0; i < arguments.length; i++) {
		if (first) {
			first = false;
		} else {
			out += ' ';
		}
		out += arguments[i];
	}
	print(out);
}

while (true) {
	log('Hello World!');
}
```
Since I used **`JERRY_SYSTEM_ALLOCATOR`**, I don't want to perform the recycling work frequently. I mask the following code:
```
static void *
jmem_heap_gc_and_alloc_block (...)
{
  ...
#if 0
  if (JERRY_CONTEXT (jmem_heap_allocated_size) + size >= JERRY_CONTEXT (jmem_heap_limit))
  {
    jmem_run_free_unused_memory_callbacks (JMEM_FREE_UNUSED_MEMORY_SEVERITY_LOW);
  }
#endif
  ...
}
```
However, an **`ERR_REF_COUNT_LIMIT`** error will occur soon.
This error is available from 2019-4-3 to all current versions.

If the code for reclaiming memory is not masked, the program execution is normal. I think that the normal execution of the program cannot depend on whether the garbage collection operation is performed frequently.

zherczeg, 2019-05-13:
As far as I remember the reference counters are 13 bit long for strings. Perhaps the first thing we need to exclude from the test is the "arguments" object. What happens with this code:

```
function log(arg) {
	var out = '';
	var first = true;
	for (var i = 0; i < 1; i++) {
		if (first) {
			first = false;
		} else {
			out += ' ';
		}
		out += arg;
	}
	print(out);
}

while (true) {
	log('Hello World!');
}
```

Is this still stops with `ERR_REF_COUNT_LIMIT` ?

dbatyai, 2019-05-13:
When using the system allocator, the first alloc_block [here](https://github.com/pando-project/jerryscript/blob/master/jerry-core/jmem/jmem-heap.c#L320) will always return a memory block, and the code after that will never run.
So the problem is, that by masking that part of the code that you mentioned, you are essentially disabling the garbage collector entirely, since that is the only remaining code that calls it.

After that it's just a matter of time to reach the ref count limit, since nothing is getting released.

acoinfo, 2019-05-13:
Oh? Garbage collection should reclaim objects with a reference count of 0. Why do gc recycle objects with valid reference counts? If my memory is very large, should I also call garbage collection frequently? 
The current code design is: uses **`JERRY_SYSTEM_ALLOCATOR`** The `jmem_heap_gc_and_alloc_block()` function will call the recycle operation every time. Is the efficiency too low?

dbatyai, 2019-05-13:
It does not recycle objects with valid reference counts. The issue is that it's not recycling anything, not even those with 0 reference, due to it not being called.

Due to this temporary objects remain in the memory indefinitely, which hold references for example to the global object, and then the reference counter will overflow fairly quickly, which results in the `ERR_REF_COUNT_LIMIT` error that you are experiencing.

Nevertheless, the behavior that you describe, that the `jmem_run_free_unused_memory_callbacks` function runs with every alloc, due to `jmem_heap_allocated_size` and `jmem_heap_allocated_size` being zero is probably a bug, and not the intended behavior, but removing the call entirely is not the appropriate fix for it.

dbatyai, 2019-05-13:
Can you please check with #2872 if you still experience the issue?

acoinfo, 2019-05-13:
Ok, understand, thank you very much.

akosthekiss, 2019-06-28:
@acoinfo Can you please get back to us with a feedback whether #2872 has solved the issue?

dbatyai, 2019-10-02:
Treating this as resolved.

closed 2019-10-02 by dbatyai