jerryscript-project/jerryscript #2868: Jerryscript performance benchmark

bryanle92, 2019-05-10:
I have noticed that Jerryscript run too slowly, especially on array manipulation. For example, this code takes about 119ms on my ARM-CortexM4 48MHz 32-bit machine:

```
var array = new Int16Array(1500) 

for (var iii = 0; iii < 1500; iii++)
{
  if (iii < 2)
  {
    array[iii] = iii
  }
  else
  {
    array[iii] = (array[iii - 1] - array[iii - 2]) * 2
  }
}
```

Build command (from commit `2d1ad149`):
```
sudo python tools/build.py  --clean --lto=off --toolchain=cmake/toolchain_mcu_stm32f4.cmake --jerry-cmdline=off --jerry-ext=off --jerry-libm=off --jerry-port-default=off --all-in-one=on --error-messages=on --mem-heap=63 --mem-stats=on --profile=my_profile --snapshot-exec=on --snapshot-save=off`
```

Profile:
```
JERRY_BUILTIN_ANNEXB=0
JERRY_BUILTIN_DATE=0
JERRY_BUILTIN_REGEXP=0
JERRY_ES2015=0
JERRY_ES2015_BUILTIN_TYPEDARRAY=1
JERRY_UNICODE_CASE_CONVERSION=0
JERRY_NUMBER_TYPE_FLOAT64=1
```

Is it the limitation of Jerryscript? Do you have any idea to improve performance in my machine?

LaszloLango, 2019-05-10:
What about normal ES5.1 arrays? What are the performance results without TypedArrays?

bryanle92, 2019-05-11:
Normal ES5.1 array is much slower than typedarray. The equivalent code of normal array takes about 185ms. Btw, typedarray doesn't work correctly if JERRY_NUMBER_TYPE_FLOAT64 is enabled. When I access any properties or methods of typedarray in Float64, the engine enter infinity loop somewhere in ecma_errol0_dtoa(). If JERRY_NUMBER_TYPE_FLOAT64 is disabled, everything is OK

```
var array = new Array(1500)

  for (var iii = 0; iii < 1500; iii++)
  {
    if (iii < 2)
    {
      array[iii] = iii
    }
    else
    {
      array[iii] = (array[iii - 1] - array[iii - 2]) * 2
    }
  }
```

LaszloLango, 2019-05-13:
I've investigated a bit with `perf` tool. Everything looks normal to me. The most of the runtime comes from the string comparisons (looking for array keys). Let's keep this issue open, maybe someone will have a good idea how to improve such cases.

akosthekiss, 2019-06-28:
> Btw, typedarray doesn't work correctly if JERRY_NUMBER_TYPE_FLOAT64 is enabled. When I access any properties or methods of typedarray in Float64, the engine enter infinity loop somewhere in ecma_errol0_dtoa().

@bryanle92 This is strange as `JERRY_NUMBER_TYPE_FLOAT64=1` is the default and that setting is well-tested. Could you open a separate issue for this if you have a (minimal?) test case that reproduces the infinite loop?

weixiongmei, 2019-06-28:
wrap it into a function and test it, you might get 2x - 6x faster or more....

bryanle92, 2019-07-01:
> @bryanle92 This is strange as `JERRY_NUMBER_TYPE_FLOAT64=1` is the default and that setting is well-tested. Could you open a separate issue for this if you have a (minimal?) test case that [...]

I also surprise that it doesn't work with `JERRY_NUMBER_TYPE_FLOAT64=1`. My test case is very simple. I tried loading this script directly and loading from a snapshot and got the same results.

`var array = new Array(1500)
 var  length = array.length`

bryanle92, 2019-07-01:
Sorry, I forgot that this issue happened with TypedArray only. So the correct test case should be:

`var array = new Int32Array(1500); var length = array.length`

Not sure that I can reproduce this issue on a desktop system but I will give it a try

dbatyai, 2019-10-04:
A few changes have landed recently that should improve the performance of TypedArrays. If you feel performance is still a problem, then please open a new issue. Closing this due to inactivity.

closed 2019-10-04 by dbatyai