jerryscript-project/jerryscript #2869: jerry-snapshot: Crashes with SIGSEV, out of bounds memory

IngwiePhoenix, 2019-05-11:
I guess a backtrace and log speak a little more than words :) Ignore the noise at the beginning, this has to do with incorrect python setup (which is very unfortunate due to Homebrew's python clashing with OS X's.)

```
Ingwie@Ingwies-MBP.fritz.box ~/W/G/j/out $ lldb ./bin/jerry-snapshot ./test.js
(lldb) target create "./bin/jerry-snapshot"
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/copy.py", line 52, in <module>
    import weakref
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
Current executable set to './bin/jerry-snapshot' (x86_64).
(lldb) settings set -- target.run-args  "./test.js"
(lldb) r
Process 63208 launched: '/Users/Ingwie/Work/Git/jerryscript/out/bin/jerry-snapshot' (x86_64)
Error: unknown command: ./test.js

Process 63208 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x314)
    frame #0: 0x0000000100003cf1 jerry-snapshot`jerry_debugger_send_log + 23
jerry-snapshot`jerry_debugger_send_log:
->  0x100003cf1 <+23>: testb  $0x1, 0x314(%rax)
    0x100003cf8 <+30>: je     0x100003d10               ; <+54>
    0x100003cfa <+32>: addb   $0x2, %bl
    0x100003cfd <+35>: movl   %r15d, %ecx
Target 0: (jerry-snapshot) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x314)
  * frame #0: 0x0000000100003cf1 jerry-snapshot`jerry_debugger_send_log + 23
    frame #1: 0x00000001000394ab jerry-snapshot`jerry_port_log + 303
    frame #2: 0x0000000100001552 jerry-snapshot`main + 254
    frame #3: 0x00007fff6b91d3d5 libdyld.dylib`start + 1
```

Contents of that test file:
```js
class A {
  mtd() {
    return 1;
  }
}

print((new A()).mtd());
```

Executing it normally works just fine.

```
$ ./bin/jerry ./test.js
Exception backtrace:
class A {
^
Script Error: SyntaxError: Primary expression expected. [line: 1, column: 1]
```

Yes, it is a syntax error, but I expected to see this when I was building a snapshot. Any idea? Thanks!

IngwiePhoenix, 2019-05-11:
Also:
```
Ingwie@Ingwies-MBP.fritz.box ~/W/G/j/out $ ./bin/jerry-snapshot generate ./test.js
Input file './test.js' (65 bytes) loaded.
fish: './bin/jerry-snapshot generate .…' terminated by signal SIGSEGV (Address boundary error)
```

IngwiePhoenix, 2019-05-11:
I realized I had not done a debug build...oops. Here is a backtrace with the debug info enabled:

```
Ingwie@Ingwies-MBP.fritz.box ~/W/G/j/out $ env PATH=/usr/bin/:$PATH lldb ./bin/jerry-snapshot generate ./test.js
(lldb) target create "./bin/jerry-snapshot"
Current executable set to './bin/jerry-snapshot' (x86_64).
(lldb) settings set -- target.run-args  "generate" "./test.js"
(lldb) r
Process 65071 launched: '/Users/Ingwie/Work/Git/jerryscript/out/bin/jerry-snapshot' (x86_64)
Input file './test.js' (40 bytes) loaded.
Process 65071 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x1e4)
    frame #0: 0x0000000100007bc0 jerry-snapshot`jerry_init(flags=JERRY_INIT_EMPTY) at jerry.c:172:3
   169 	jerry_init (jerry_init_flag_t flags) /**< combination of Jerry flags */
   170 	{
   171 	  /* This function cannot be called twice unless jerry_cleanup is called. */
-> 172 	  JERRY_ASSERT (!(JERRY_CONTEXT (status_flags) & ECMA_STATUS_API_AVAILABLE));
   173 	
   174 	  /* Zero out all non-external members. */
   175 	  memset (&JERRY_CONTEXT (JERRY_CONTEXT_FIRST_MEMBER), 0,
Target 0: (jerry-snapshot) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x1e4)
  * frame #0: 0x0000000100007bc0 jerry-snapshot`jerry_init(flags=JERRY_INIT_EMPTY) at jerry.c:172:3
    frame #1: 0x0000000100002169 jerry-snapshot`process_generate(cli_state_p=0x00007ffeefbff828, argc=3, prog_name_p="/Users/Ingwie/Work/Git/jerryscript/out/bin/jerry-snapshot") at main-unix-snapshot.c:297:3
    frame #2: 0x00000001000015fd jerry-snapshot`main(argc=3, argv=0x00007ffeefbff880) at main-unix-snapshot.c:774:18
    frame #3: 0x00007fff6b91d3d5 libdyld.dylib`start + 1
```

IngwiePhoenix, 2019-05-11:
And because it might be important, here is also my CMakeCache.

<details>
<summary>CMakeCache.txt</summary>
# This is the CMakeCache file.
# For build in directory: /Users/Ingwie/Work/Git/jerryscript/out
# It was generated by CMake: /usr/local/Cellar/cmake/3.14.3/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Build shared libraries?
BUILD_SHARED_LIBS:BOOL=OFF

//Path to a program.
CMAKE_AR:FILEPATH=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ar

//Choose the type of build, options are: None Debug Release RelWithDebInfo
// MinSizeRel ...
CMAKE_BUILD_TYPE:STRING=Debug

//C compiler
CMAKE_C_COMPILER:FILEPATH=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc

//Flags used by the C compiler during all build types.
CMAKE_C_FLAGS:STRING=

//Flags used by the C compiler during DEBUG builds.
CMAKE_C_FLAGS_DEBUG:STRING=-g

//Flags used by the C compiler during MINSIZEREL builds.
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the C compiler during RELEASE builds.
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG

//Flags used by the C compiler during RELWITHDEBINFO builds.
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG

//Executable file format
CMAKE_EXECUTABLE_FORMAT:STRING=MACHO

//Flags used by the linker during all build types.
CMAKE_EXE_LINKER_FLAGS:STRING=

//Flags used by the linker during DEBUG builds.
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=

//Flags used by the linker during MINSIZEREL builds.
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=

//Flags used by the linker during RELEASE builds.
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=

//Flags used by the linker during RELWITHDEBINFO builds.
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=

//Enable/Disable output of compile commands during generation.
CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=OFF

//Path to a program.
[... 377 lines cut ...]
//MODIFIED property for variable: FEATURE_DEBUGGER
FEATURE_DEBUGGER-MODIFIED:INTERNAL=ON
//MODIFIED property for variable: FEATURE_ERROR_MESSAGES
FEATURE_ERROR_MESSAGES-MODIFIED:INTERNAL=ON
//MODIFIED property for variable: FEATURE_EXTERNAL_CONTEXT
FEATURE_EXTERNAL_CONTEXT-MODIFIED:INTERNAL=ON
//MODIFIED property for variable: FEATURE_LINE_INFO
FEATURE_LINE_INFO-MODIFIED:INTERNAL=ON
//MODIFIED property for variable: FEATURE_LOGGING
FEATURE_LOGGING-MODIFIED:INTERNAL=ON
//MODIFIED property for variable: FEATURE_MEM_STATS
FEATURE_MEM_STATS-MODIFIED:INTERNAL=ON
//MODIFIED property for variable: FEATURE_SNAPSHOT_EXEC
FEATURE_SNAPSHOT_EXEC-MODIFIED:INTERNAL=ON
//MODIFIED property for variable: FEATURE_SNAPSHOT_SAVE
FEATURE_SNAPSHOT_SAVE-MODIFIED:INTERNAL=ON
//Have include time.h
HAVE_TIME_H:INTERNAL=1
//Test HAVE_TM_GMTOFF
HAVE_TM_GMTOFF:INTERNAL=1
//Have include unistd.h
HAVE_UNISTD_H:INTERNAL=1
//MODIFIED property for variable: JERRY_CMDLINE_SNAPSHOT
JERRY_CMDLINE_SNAPSHOT-MODIFIED:INTERNAL=ON
</details>

IngwiePhoenix, 2019-05-12:
Turns out this behaviour only shows with `FEATURE_EXTERNAL_CONTEXT` being enabled.

Is this supposed to happen?

rerobika, 2019-05-13:
@IngwiePhoenix Thanks for the report the error seems valid. The usage of `FEATURE_EXTERNAL_CONTEXT` was unsupported in the snapshot tool. I made a PR to resolve this error, please check it.

PR #2870 (2019-05-13): Support external context in the snapshot tool
This patch fixes #2869.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 76a1fc56 (2019-05-13): Support external context in the snapshot tool (#2870)

This patch fixes #2869.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2019-05-13 by LaszloLango