jerryscript-project/jerryscript #2876: Right way to get a string from JerryScript?

IngwiePhoenix, 2019-05-17:
I am working on writing wrappers for the [rlutil](https://github.com/tapio/rlutil) library, and right now I am a bit lost on the way to convert `jerry_value_t`s into `char*`s.

```c
static jerry_value_t
rlutil_setConsoleTitle(
    const jerry_value_t func_value,
    const jerry_value_t this_value,
    const jerry_value_t *arg_p,
    const jerry_value_t arg_cnt
) {
    if(arg_cnt != 1) {
        // handle eerror
    }

    jerry_size_t len = jerry_get_string_size();
    char str[len + 1];
    jerry_string_to_char_buffer(arg_p[0], &str, len);

    setConsoleTitle(str);

    return jerry_create_undefined();
}
```

I haven't had a opportunity to test just yet - writing this in my break time :). But, is this the correct way of doing this, and is there possibly a better way?

Thanks!

LaszloLango, 2019-05-17:
@IngwiePhoenix looks good to me.
Note: `str[]` will be local variable, so it won't be initialzed to zero. I'd add `str[len] = 0;` to your code to ensure it will be zero terminated.

In general, [`jerry_string_to_char_buffer`](http://jerryscript.net/api-reference/#jerry_string_to_char_buffer) is used for CESU8 encoded strings and [`jerry_string_to_utf8_char_buffer`](http://jerryscript.net/api-reference/#jerry_string_to_utf8_char_buffer) is used for UTF8 encoded strings. You can validate the `jerry_value_t` with `jerry_is_valid_utf8_string` or `jerry_is_valid_cesu8_string`. Furthermore you can compare the return value of `jerry_string_to_char_buffer` to `jerry_get_string_size`. They must be equal if the call was successful and all characters were copied to the buffer.

http://jerryscript.net/api-reference/#jerry_string_to_utf8_char_buffer

LaszloLango, 2019-05-17:
Note 2: UTF8 and CESU8 strings can contain zeros as valid characters, so it won't be zero terminated as you expect in your code. (Except if you guarantee it yourself)

IngwiePhoenix, 2019-05-17:
> @IngwiePhoenix looks good to me.
> Note: `str[]` will be local variable, so it won't be initialzed to zero. I'd add `str[len] = 0;` to your code to ensure it will be zero terminated.

Oh! Good point :) Totally forgot that one...
 
> In general, [`jerry_string_to_char_buffer`](http://jerryscript.net/api-reference/#jerry_string_to_char_buffer) is used for CESU8 encoded strings and [...]

Ah, that might not be important for a console title, but for larger strings for sure. Thank you for pointing this out, useful thing to `assert()` for in some cases!

> Note 2: UTF8 and CESU8 strings can contain zeros as valid characters, so it won't be zero terminated as you expect in your code. (Except if you guarantee it yourself)

I am currently thinking about a fast and efficient way to check if that buffer contains zeros or not - but for right now, this is not a big concern that I have. Still, good to have that in mind. :)

Thank you very much, @LaszloLango !

closed 2019-05-17 by IngwiePhoenix

zherczeg, 2019-05-17:
One more note: this only works with strings. If you have non-string values (numbers, objects, etc.) you should use `jerry_value_to_string` first.