jerryscript-project/jerryscript #2877: Suggestion: jerryscript-ext/module.h - Improve built-in modules

IngwiePhoenix, 2019-05-17:
I would like to suggest a set of useful macros, for the case that someone is building modules right into the host application versus loading them dynamicaly.

The idea behind a macro like `JERRYX_NATIVE_MODULE_EXTERN(module_name)` would be, that a developer may be curating a list of built-in modules, through which they may iterate within the initialization of a program:

```c
jerryx_native_module_t internal_modules[] = {
  _foo_definition,
  _bar_definition,
  NULL
};
for(int i = 0; internal_modules[i] != NULL; i++) {
  jerryx_native_module_register(internal_modules[i]);
}
```

Currently, the developer has to actually be aware of the definition of `JERRYX_NATIVE_MODULE(...)` in order to do this. Having a set of helper macros that simplify the usage of this method would be quite helpful therefore. The macros would simply help to get the names right; to retrive the on-resolve callback, or the respective register or unregister function, if required.

What do you think? :) 

Kind regards,
Ingwie

LaszloLango, 2019-05-17:
CC: @gabrielschulhof

gabrielschulhof, 2019-05-17:
@IngwiePhoenix absolutely. I'm thinking, as you said, `JERRYX_NATIVE_MODULE_EXTERN()` and
```C
#define JERRYX_NATIVE_MODULE_DEFINITION(module_name) \
  _ ## module_name ## _definition
```

and then you could do
```C
jerryx_native_module_t internal_modules[] = {
  JERRYX_NATIVE_MODULE_DEFINITION(foo),
  JERRYX_NATIVE_MODULE_DEFINITION(bar)
};
```

Just gotta keep in mind that this would constrain us in the future such that the `JERRYX_NATIVE_MODULE_DEFINITION()` macro must refer to something you can mention in an array initialization.

Would you like to submit a PR?

IngwiePhoenix, 2019-05-17:
Sure! I will do so in a little - after finishing a conceptual C++ wrapper for JerryScript. The project I am working on has me use a lot of C++ code as well as a lot of JS code - so, I decided to write a very simple and thin wrapper over JerryScript to make things a tiny bit friendlier. :) I'll make sure to let you guys know once I have a working example, too!

I'll ping this thread when I submitted the PR.

IngwiePhoenix, 2019-05-17:
#2878 :)

gabrielschulhof, 2019-05-24:
@IngwiePhoenix actually, I think your use case can be addressed with the existing code as follows:

```C
void foo_register();
void bar_register();

void (*internal_modules)(void)[] = {
  foo_register,
  bar_register,
  NULL
};
for(int i = 0; internal_modules[i] != NULL; i++) {
  internal_modules[i]();
}
```

This is actually what the module test does, albeit without first wrapping the module registration functions into an array. The fact is though that with `ENABLE_INIT_FINI` turned on the C library basically runs the above code for you without your having to write it.

Sorry I led you on a wild goose chase with exposing the module definition structure! I haven't concentrated entirely on the problem and I had forgotten about the importance of keeping data symbols local.

rerobika, 2019-10-18:
Closing the issue after #2878, please reopen it anytime when it becomes necessary.

closed 2019-10-18 by rerobika