jerryscript-project/jerryscript #2879: Provide import(String):Promise

IngwiePhoenix, 2019-05-17:
Apparently, a recent ECMA standard specifies `import(module: String): Promise` as a dynamic method of loading and processing modules.

Are there plans for implementing this?

galpeter, 2019-05-17:
I've been looking around to find this in the standard, could you provide a point where this is described?

IngwiePhoenix, 2019-05-17:
- http://2ality.com/2017/01/import-operator.html
- https://developers.google.com/web/updates/2017/11/dynamic-import
- https://github.com/tc39/proposal-dynamic-import (a couple of links to the draft and standard might be here)
  * https://tc39.github.io/proposal-dynamic-import/

This is what I found on first glance.

akosthekiss, 2019-06-28:
I mark this as a feature request but I'm also closing the issue now. This feature is still a stage 4 draft proposal for JavaScript as of date. This means that it is not even in the standard yet. JerryScript is ES5.1 compatible, and plans to reach ES6/2015 compatibility in the future. But the project cannot take on the implementation of draft proposals at the moment.

closed 2019-06-28 by akosthekiss