jerryscript-project/jerryscript #2880: Assigning a native pointer to an object - individually.

IngwiePhoenix, 2019-05-17:
As part of my project, I need to expose some C++ APIs to JerryScript. Now, the best way to actually do that is by being able to tag the native instance onto the object created. Imagine this class for example:

```c++
class Foo {
public:
  Foo();
  ~Foo();
  int someMethod();
};
```

Now, what I guess I have to do, is to create a `jerry_native_object_info_t`.

```c
void foo_free(void* ptr) {
  Foo* foo_ptr = (Foo*)ptr;
  delete foo_ptr; // trigger deallocation and deconstruction
}
jerry_object_native_info_t foo_info = {
  .free_cb = foo_free
};
```

So far, so good. But from what I figured, I would do something like this:
```c
jerry_value_t obj = jerry_create_object();
Foo* foo_ptr = new Foo();
jerry_set_object_native_pointer(obj, (void*)foo_ptr, &foo_info);
```

And get this pointer in a method like this:
```c
void* ptr;
if(jerry_get_object_native_pointer(obj, &ptr, &foo_info)) {
  Foo* foo_ptr = (void*)ptr;
}
```

So far, this bit I did understand. But what if I want to have the instance of `Foo` tagged on an instance object? Do I define that in an `Object.constructor`? Is it save that - even when I have multiple `Foo`s around, that only the instance tagged to the specific object will be sent to the `free_cb` once the GC decides to delete it? (And on a side note: how do I actually define a constructor in C?)

Also, can I tag any kind of pointer in? Take for instance: I am working on a C++ wrapper, and I would like to reduce this:
```c
static jerry_value_t
function(
  const jerry_value_t func_value,
  const jerry_value_t this_value,
  const jerry_value_t *args_p,
  const jerry_value_t args_cnt
);
```

Into this:
```c++
Value function(Context ctx);
```

The idea I had, was to use a generic wrapper object which would construct a function off something like the following, and just take the returned value and return it like usual:
```c++
typedef Value (*jerry_cxx_handler_t(Context);
jerry_object_native_info_t dummy_info = {
  .free_cb = NULL
};
static jerry_value_t
cxx_wrapper(
  const jerry_value_t func_value,
  const jerry_value_t this_value,
  const jerry_value_t *args_p,
  const jerry_value_t args_cnt
) {
  void* ptr;
  if(! jerry_get_object_native_pointer(func_value, &ptr, &dummy_info)) { /* handle error */ }
  jerry_cxx_handler_t handler = (jerry_cxx_handler_t)ptr;
  Context ctx(func_value, this_value, args_p, args_cnt);
  Value cxx_rt = handler(ctx);
  jerry_value_t rt = cxx_rt.raw();
  return rt;
  // ctx and cxx_rt should auto-destruct and release their copy of the value properly.
}
```

Now, what I am curios about is: When constructing the `jerry_value_t` with the native pointer set to a `jerry_cxx_handler_t`, is it ensured that only _this_ `func_value` has exactly _this_ function pointer assigned at any time?

Thank you very much. :)

LaszloLango, 2019-05-17:
FYI: @fbmrk did a similar code generator for IoT.js: https://github.com/pando-project/iotjs/pull/1756
IoT.js is a Node.js like JavaScript framework built on top of JerryScript. I don't know if it can be used in your use case or can it be modified to use JerryScript only.

IngwiePhoenix, 2019-05-22:
@LaszloLango I looked into it, and that might just be a little bit more than I actually want. All that I really want is to find out if assigning a pointer to a `jerry_value_t` of type `Function` will always stay the same for this exact object.

Imagine this:
```js
function foo() {}

// This is made-up, but you get the idea.
var ptr = <...some native pointer...>
Object.setNativePointer(foo, ptr)
```

Now, whenever I retrive `foo`, and extract it's native pointer, am I guaranteed to always get exactly that pointer back - or may the assigned pointer appear on any `jerry_value_t` of type `Function` once assigned?

LaszloLango, 2019-05-22:
Yes, it will get always the same pointer for a certain value. The pointer is assigned to the value and not to the type of the value.

IngwiePhoenix, 2019-05-22:
Perfect :) I just wanted to make sure before I invest myself into something, only to end up running into quite a wall.

Thank you very much - this is very, very helpful!

closed 2019-05-22 by IngwiePhoenix