jerryscript-project/jerryscript #2883: Curios/Question: Exchanging implementations

IngwiePhoenix, 2019-05-22:
This is more of a curios question in the way of it being "theoreticaly possible or not".

Basically, I was wondering if I could slightly improve JerryScript's performance by swapping out certain parts of it - like it's JSON parser with something like [simdjson](https://github.com/lemire/simdjson), it's string handling with [rapidstring](https://github.com/boyerjohn/rapidstring), RegExp with [re2](https://github.com/google/re2) and it's compiler and VM with [asmjit](https://github.com/asmjit/asmjit).

Would any of this be possible? I doubt that just re-implementing the API won't do, forking the entire project seems more likely for some of the things. But what is most interesting to me is improving JerryScript's performance by using a JIT instead of an interpreter.

What do you think of this?

LaszloLango, 2019-05-22:
IMHO you should fork JerryScript if you want to do things like this. This would be a huge rework. Change of the RegExp and JSON engines might be simple, but changing the VM doesn't.

galpeter, 2019-05-23:
An important aspect for the mainline JerryScript is the code size (compiled binary size) and the memory usage/requirement. Still, as previously said changing the JSON/RegExp parts should would be easier than the VM (or adding JIT)

akosthekiss, 2019-06-28:
Closing as the question was answered without further discussion.

closed 2019-06-28 by akosthekiss